RedApparat/Fotoapparat #2: Samsung Galaxy S5. Camera is not starting

shivam2702, 2017-04-24:
04-24 15:45:32.639 19867-19894/io.fotoapparat.sample I/OpenGLRenderer: Initialized EGL, version 1.4
04-24 15:45:32.649 19867-19893/io.fotoapparat.sample W/ArrayUtils: Ignoring invalid value emboss
04-24 15:45:32.649 19867-19893/io.fotoapparat.sample W/ArrayUtils: Ignoring invalid value sketch
04-24 15:45:32.649 19867-19893/io.fotoapparat.sample W/ArrayUtils: Ignoring invalid value neon
04-24 15:45:32.649 19867-19893/io.fotoapparat.sample I/CameraManager: Using legacy camera HAL.
04-24 15:45:32.669 19867-19894/io.fotoapparat.sample I/OpenGLRenderer: HWUI protection enabled for context ,  &this =0xaf722088 ,&mEglDisplay = 1 , &mEglConfig = 8 
04-24 15:45:32.679 19867-19894/io.fotoapparat.sample D/OpenGLRenderer: Enabling debug mode 0
04-24 15:45:32.749 19867-19867/io.fotoapparat.sample I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@2f1180cc time:32426103
04-24 15:45:32.959 19867-19893/io.fotoapparat.sample W/ArrayUtils: Ignoring invalid value emboss
04-24 15:45:32.959 19867-19893/io.fotoapparat.sample W/ArrayUtils: Ignoring invalid value sketch
04-24 15:45:32.959 19867-19893/io.fotoapparat.sample W/ArrayUtils: Ignoring invalid value neon
04-24 15:45:32.979 19867-19893/io.fotoapparat.sample D/Fotoapparat: getCapabilities
04-24 15:45:32.979 19867-19893/io.fotoapparat.sample D/Fotoapparat: updateParameters
04-24 15:45:32.979 19867-19893/io.fotoapparat.sample D/Fotoapparat: setDisplayOrientation
04-24 15:45:32.979 19867-19893/io.fotoapparat.sample D/Fotoapparat: getRendererParameters
04-24 15:45:32.979 19867-19893/io.fotoapparat.sample D/Fotoapparat: setDisplaySurface
04-24 15:45:32.979 19867-19893/io.fotoapparat.sample D/Fotoapparat: startPreview
04-24 15:45:32.989 19867-19893/io.fotoapparat.sample E/CameraDevice-0-LE: Surface with size (w=1440, h=1920) and format 0x1 is not valid, size not in valid set: [1920x1080, 1440x1080, 1280x720, 1056x864, 960x720, 800x480, 720x480, 640x480, 352x288, 320x240, 176x144]
04-24 15:45:32.989 19867-19893/io.fotoapparat.sample W/CameraDevice-JV-0: Stream configuration failed
04-24 15:45:32.989 19867-19893/io.fotoapparat.sample E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed

dmitry-zaitsev, 2017-04-26:
@shivam2702 thanks a lot for reporting it. I am trying to reproduce it, but with no luck. Could you please check which preview sizes you are receiving in `io.fotoapparat.hardware.Capabilities` constructor?

dmitry-zaitsev, 2017-04-26:
Also would be nice to know which OS version you are using.

dmitry-zaitsev, 2017-04-29:
After many hours of trying to reproduce that using other devices, Samsung Remote Lab and finally actual Samsung S5 I was not able to reproduce that. I am closing this issue until there will be more information.

closed 2017-04-29 by dmitry-zaitsev

shivam2702, 2017-05-01:
Capabilities: previewSizes [Size{width=1440, height=1080}, Size{width=176, height=144}, Size{width=352, height=288}, Size{width=640, height=480}, Size{width=320, height=240}, Size{width=800, height=480}, Size{width=960, height=720}, Size{width=1056, height=864}, Size{width=1920, height=1080}, Size{width=1280, height=720}, Size{width=720, height=480}]

shivam2702, 2017-05-01:
W/LegacyRequestMapper: convertRequestMetadata - control.awbRegions setting is not supported, ignoring value
W/LegacyRequestMapper: Only received metering rectangles with weight 0.

Sometimes the camera previews but not with the appropriate resolution and throws this warning.

Thank you for you quick response and wonderful support.