RedApparat/Fotoapparat #3: Front Camera Issue (Not Capturing)

shivam2702, 2017-04-24:
In some devices i faced an issue that its not capturing the Photo by the front camera. 

Ex: One Plus 2 etc...

Sorry for no logs as its not showing any.

Diolor, 2017-04-24:
Hi @shivam2702, was it API 21+ ? 
Also did any of the following was the case: rotate the activity, had another take request before, or low light?

shivam2702, 2017-04-24:
API Level was 23 (Android M)
No everything was fine best light first request and portrait.

dmitry-zaitsev, 2017-04-24:
We are able to reproduce the issue. Fix will come soon.

Diolor, 2017-05-01:
This issue is now fixed 👍

shivam2702, 2017-05-03:
Thanks 🥇  you are awesome....

dmitry-zaitsev, 2017-05-09:
Still an issue for Samsung S7 Edge. Did not try on Moto G3 yet.

Diolor, 2017-05-09:
Default FA configuration in sample app? Sorry cannot reproduce in my 6P

dmitry-zaitsev, 2017-05-09:
@Diolor issue is in `ImageCaptureAction#getPhoto` - for some reason `countDownLatch` is never actually counted down. I suspect that it might be because image becomes available before listener is registered and therefore `onImageAvailable` is not called.

dmitry-zaitsev, 2017-05-09:
Fixed now.

closed 2017-05-09 by dmitry-zaitsev