RedApparat/Fotoapparat #5: Image squeezed up after click (CAMERA API 2)

shivam2702, 2017-04-26:
Device : OnePlus 2
API level : 23 
Image links : before capture https://drive.google.com/file/d/0B7Rq1Vju8a84c0FMbVpNRTg3d28/view?usp=drivesdk

After capture 
https://drive.google.com/file/d/0B7Rq1Vju8a84SXNZdm9xVlFkbmM/view?usp=drivesdk

dmitry-zaitsev, 2017-04-26:
@shivam2702 I am trying to figure out whether it is a preview which is squeezed or a resulting image (since we don't have a Mac Pro at hand :-) ). Could you please instead take a photo of this nice square?
https://img.clipartfest.com/6a12b006598ff323cf6270ae75399da9_open-square_1000-1000.png

It would be easy to verify which photo is squished and by how much.

Diolor, 2017-05-01:
Thanks @shivam2702!
I was able to reproduce a malformed preview after taking a picture with a front face fixed camera in camera 2. No need of pictures of mac pros or squares :)

shivam2702, 2017-05-02:
Hi,
Sorry I was lil busy and i found the reason behind that you can see in the screenshots that's it's not stretching out but capturing in wrong dimensions  
[image]
[image]

dmitry-zaitsev, 2017-05-02:
@Diolor is that the same issue which you were able to reproduce?

Diolor, 2017-05-02:
I reproduced with front camera, but I expect to be the same

dmitry-zaitsev, 2017-05-02:
Then I think this is not actually a bug. Since preview is fit into the view in a such way that it covers the whole screen we have to cut away some parts of it. When we are taking the photo, we don't care about that and display result as is. Hence image is perceived differently.

What we could do to solve the problem is to allow users to fit image inside the screen. I think this should be as simple as modifying `CameraView`.

shivam2702, 2017-05-05:
can you please guide me how to do that @dmitry-zaitsev

dmitry-zaitsev, 2017-05-06:
@shivam2702 if you feel like exploring, you can try to change line 125 in `TextureRendererView` from `Math.max` to `Math.min` and see if that helps. Otherwise I will do that as soon as I'm back from my vacation (which is in a couple of days).

shivam2702, 2017-05-08:
Thx @dmitry-zaitsev  enjoy the vacations buddy 👍

shivam2702, 2017-05-09:
I tried it and i have noticed that 

// THIS GIVE FULL SIZE  AND BEST PREVIEW
@Override
    protected void onLayout(boolean changed, int left, int top, int right, int bottom) {
        if (previewSize == null) {
            super.onLayout(changed, left, top, right, bottom);
            return;
        }

        final float scale = Math.max(
                getMeasuredWidth() / (float) previewSize.width,
                getMeasuredHeight() / (float) previewSize.height
        );

        final int width = (int) (previewSize.width * scale);
        final int height = (int) (previewSize.height * scale);

        final int extraX = Math.max(0, width - getMeasuredWidth());
        final int extraY = Math.max(0, height - getMeasuredHeight());

        getChildAt(0).layout(
                -extraX / 2,
                -extraY / 2,
                width - (extraX / 2),
                height - (extraY / 2)
        );
    }

// AFTER USING MIN IT CHANGES THE ASPECT RATIO
final float scale = Math.min(
                getMeasuredWidth() / (float) previewSize.width,
                getMeasuredHeight() / (float) previewSize.height
        );

//AFTER USING MIN EVERY WHERE HERE THE ASPECT RATIO BECOMES OK BUT BLACK PORTION IS AUTOMATICALLY ADDED TO THE PREVIEW BUT I AM LOOKING FOR FULL SCREEN PREVIEW AND FULL SCREEN PHOTO.

@Override
    protected void onLayout(boolean changed, int left, int top, int right, int bottom) {
        if (previewSize == null) {
            super.onLayout(changed, left, top, right, bottom);
            return;
        }

        final float scale = Math.min(
                getMeasuredWidth() / (float) previewSize.width,
                getMeasuredHeight() / (float) previewSize.height
        );

        final int width = (int) (previewSize.width * scale);
        final int height = (int) (previewSize.height * scale);

        final int extraX = Math.min(0, width - getMeasuredWidth());
        final int extraY = Math.min(0, height - getMeasuredHeight());

        getChildAt(0).layout(
                -extraX / 2,
                -extraY / 2,
                width - (extraX / 2),
                height - (extraY / 2)
        );
    }

Diolor, 2017-05-09:
I think I understand. Leave the preview view as is and add `android:scaleType="centerCrop"` in your photo result ImageView.

Notice: this will "fit" the photo in the image view and give you a  fullscreen effect but due to aspect ratio reasons but "some of the photo" will not be there. :)

dmitry-zaitsev, 2017-05-09:
@shivam2702 please try what @Diolor suggested, I believe that is the root of your issue.

shivam2702, 2017-05-10:
I don't think its a solution as now the preview will be fine as wanted but at the end of the task the output image is still gives the extra spaces.
I request you to please reopen the issue

Diolor, 2017-05-10:
@shivam2702 If you added the code 2 comments above then it will not give any spaces. If you added it and still have the issue, please ensure your ImageView is fullscreen and without margins/paddings

dmitry-zaitsev, 2017-05-10:
@shivam2702 could you please also paste here your XML layout?

shivam2702, 2017-05-10:
Thx i have fixed the issue by cropping the left & right portion.
👍

closed 2017-05-10 by shivam2702