RedApparat/Fotoapparat #7: Videoapparat?

ArtworkAD, 2017-05-01:
Hey guys, thanks for this awesome library. I just wanted to check out with you, whether you plan to implement video capabilities. Is it planed or already implemented/possible to implement?

dmitry-zaitsev, 2017-05-01:
Hey, thanks for the interest! 

Short answer - no, it is not implemented yet. But it is possible to do so.

One of the reasons why we started doing Fotoapparat is because after few years of struggle with Camera1 we felt that we have enough experience to do things "the right way". Now, let me be honest - I don't feel the same way when it comes to video recording as I never worked with it. All in all we would love to do Videoapparat, but we feel that we are not ready for it yet.

Yet, since it is an open source library we are always opened for contributors ;)

ArtworkAD, 2017-05-01:
@dmitry-zaitsev thanks for the info 👍

eygraber, 2017-09-19:
I would leave this issue open so that uses can add +1s to it if they want to see the feature. If enough people want it, that might entice someone to pick it up.

dmitry-zaitsev, 2017-09-19:
@eygraber fair enough, I will reopen it.

idish, 2017-10-20:
Of course that would be great, as there is lack of libraries that are actually doing good job and have maintenance through years.

MuhammedRefaat, 2017-11-07:
Yeah it would be great if there is video recording capabilities too

Diolor, 2017-11-11:
Hi guys, we are currently examining weather or not to work on Videoapparat. We are looking the demand over how many of you need it.

If you are interested in it, I would kindly like to ask you to enter you email here: [http://fotoapparat.io](http://fotoapparat.io) so we can get an overview

Diolor, 2018-02-12:
Hey all,

Today, we are happy to announce the first Videoapparat version! 🎉

It allows both taking pictures & recording videos with minimal configuration. 
We also used a simple API following existing Fotoapparat patterns.

Note however that the development was not trivial. For that reason we decided to make it a premium library. 
Fotoapparat will always stay free and open source.

So see the [working demo](https://github.com/Fotoapparat/Videoapparat), find out more details in [our website](http://fotoapparat.io/) or [contact us to buy now](mailto:fotoaparrat.team@gmail.com).

Best,
Dio

friendoye, 2018-02-12:
2.000€? 👏 👏 👏

valterh4ck3r, 2018-02-26:
2k ?  or 2 euros ?

ezaquarii, 2018-02-26:
2k is pretty decent price by commercial standards. I'm glad somebody finally did the work. Camera API on Android is unworkable from financial point of view.

idish, 2018-02-26:
Well, If it is indeed 2k euros, I'd change the price description to 2,000€ instead of 2.000€. (i.e comma instead of a dot).
It is pretty confusing, and there's a huge difference between 2k euros and 2 euros.

Diolor, 2018-02-26:
Thanks for your feedback @idish. The price is in Euro and afaik, all Eurozone countries use the dot as thousands delimiter. We will use something else to avoid further confusion.

So bottom line:
price is 2000€ (or around 2,460 US$)

BrunoTrovoN26, 2019-04-03:
Is this project alive?

imohchard, 2019-04-09:
I am wondering if its still alive.

closed 2023-11-28 by artworkad