RedApparat/Fotoapparat #13: Aspect ratio of the preview is broken sometimes while using Camera2

dmitry-zaitsev, 2017-05-02:
Device: S7 Edge

This screen is actually 30" and certainly not square :)

[image]

Diolor, 2017-05-02:
I expect this to be duplicate of #5. Will leave open for now.

dmitry-zaitsev, 2017-05-02:
It is not duplicate, because in #5 aspect ratio is actually correct (as you can see from rectangle).

Diolor, 2017-05-04:
Not after the click.
Looks the same issue because on start preview there is a change of this stretching but always after a take photo (which call start preview again) as in #5.

Diolor, 2017-05-07:
Should be fixed. @dmitry-zaitsev please confirm by closing the ticket:)

dmitry-zaitsev, 2017-05-09:
@Diolor no, issue is still here :(

dmitry-zaitsev, 2017-05-18:
As discussed, let's move it to 1.1.0, since we are going to disable Camera2 for now.

Diolor, 2017-09-30:
I will close this v2 issue for now.

closed 2017-09-30 by Diolor