RedApparat/Fotoapparat #17: Photo is not rotated correctly in Camera1

Diolor, 2017-05-09:
Device: Nexus 6P

Potentially due to sensor orientation degrees as camera 2 with other devices (  #12 ).

[image]

dmitry-zaitsev, 2017-05-09:
@Diolor could you please tell me which parameters do you receive in `OrientationUtils#computeImageOrientation` and what is the output of the method?

Diolor, 2017-05-10:
For Pixel XL:
Front camera:
`screenRotationDegrees : 0, cameraRotationDegrees: 270, cameraIsMirrored: true`
Output is at 90.
Back camera:
`screenRotationDegrees : 0, cameraRotationDegrees: 90, cameraIsMirrored: false`
Output is at 270.

For Nexus 6P:
Back/Front camera:
`screenRotationDegrees : 0, cameraRotationDegrees: 90, cameraIsMirrored: false/true`
Output is both at 270.

dmitry-zaitsev, 2017-05-10:
All good, I can reproduce it on my Nexus 5X.

dmitry-zaitsev, 2017-05-10:
@Diolor could you please check if `fix/issue-17` branch works for you with Camera1 and Camera2?

Diolor, 2017-05-11:
both look ok on my 6P

dmitry-zaitsev, 2017-05-11:
Fixed now.

closed 2017-05-11 by dmitry-zaitsev