RedApparat/Fotoapparat #19: Samsung: taking a picture never resolves the pending result

Draftdog, 2017-05-09:
I.e. the 'whenAvailable(..)' callback is never invoked for some reason.
Same story when I do the same thing with the synchronous .await()
I'm using the same code as you have on your main github page.

Tested with:
Samsung S6 Edge (Android 6.0.1): Failed
Samsung S8 (Android 7.0): Failed
Samsung S5 (6.0.1): Failed
LG Nexus 5X (7.1.2): Success

Will test Google Pixel C Tablet tomorrow.

I never waited more than 15 minutes for the callback to be invoked.

I was replacing my old camera-handling code with Fotoapparat while updating an existing project, the old code still works fine so I doubt it's a hardware issue.

dmitry-zaitsev, 2017-05-10:
@Draftdog thanks for reporting. I fixed this issue one hour before you reported it, so it could be that fix was not yet deployed at the time you tried the library. Could you please try again with `-SNAPSHOT` version?

Draftdog, 2017-05-10:
Hi @dmitry-zaitsev,
Thanks for the fast reply, the update has indeed fixed the issue!
Cheers

closed 2017-05-10 by Draftdog