RedApparat/Fotoapparat #20: How to switch between camera1 and camera2? How to switch camera from front to back or vice versa?

shivam2702, 2017-05-10:


dmitry-zaitsev, 2017-05-10:
We do not yet have a defined API for switching between the cameras, but it is in the works and has a high priority (right below all the bugs). For now as a temporary solution I suggest creating 2 instances of `Fotoapparat` - one for front camera and one for back camera.

shivam2702, 2017-05-15:
ohk Thanks and now why don't you push a stable version?
I think current snapshot version is one of the best and very well stable .

Diolor, 2017-05-15:
@shivam2702 There are few issues we still want to resolve. Especially the aspect ration in camera2. You can check what's left here: https://github.com/Fotoapparat/Fotoapparat/milestone/1

Diolor, 2017-05-21:
Fixed on #27

closed 2017-05-21 by Diolor

dmitry-zaitsev, 2017-05-22:
@shivam2702 sample app now allows you to switch between the cameras. You can use it as an example.

shivam2702, 2017-05-22:
Thank you buddy. 
and Congratulations for the first release...