RedApparat/Fotoapparat #26: With camera1, getting "IllegalStateException: Camera error code: 1"

hagabaka, 2017-05-19:
Since d8fad94, my app now crashes with this backtrace:
```
java.lang.IllegalStateException: Camera error code: 1
                                                                            at io.fotoapparat.hardware.v1.Camera1$1.onError(Camera1.java:73)
                                                                            at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146)
 at android.os.Handler.dispatchMessage(Handler.java:102)
 at android.os.Looper.loop(Looper.java:154)
 at android.app.ActivityThread.main(ActivityThread.java:6121)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```
When `camera2` was the default, the app did not have this problem.

dmitry-zaitsev, 2017-05-19:
Could you please also attach all logs with `Fotoapparat` tag? Or just all logs.

You also have an option to switch back to Camera2, if you like. That's how you can do it:

```java
builder
    .cameraProvider(v2(context))
```

hagabaka, 2017-05-19:
Here's the full log in Android Studio if I select "Show only selected application". 
```
05-19 15:57:14.943 4343-4343/? I/art: Late-enabling -Xcheck:jni
05-19 15:57:15.083 4343-4343/? W/System: ClassLoader referenced unknown path: /data/app/com.my.app/lib/arm64
05-19 15:57:15.102 4343-4343/? I/InstantRun: starting instant run server: is main process
05-19 15:57:15.132 4343-4350/? I/art: Debugger is no longer active
05-19 15:57:15.132 4343-4350/? I/art: Starting a blocking GC Instrumentation
05-19 15:57:15.165 4343-4343/? W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
05-19 15:57:15.428 4343-4364/? I/Adreno: QUALCOMM build                   : 6818200, Idb2b4cb785
                                         Build Date                       : 11/17/16
                                         OpenGL ES Shader Compiler Version: XE031.09.00.04
                                         Local Branch                     : N25
                                         Remote Branch                    : 
                                         Remote Branch                    : 
                                         Reconstruct Branch               : 
05-19 15:57:15.431 4343-4364/? I/OpenGLRenderer: Initialized EGL, version 1.4
05-19 15:57:15.431 4343-4364/? D/OpenGLRenderer: Swap behavior 1
05-19 15:57:15.471 4343-4343/? W/art: Before Android 4.1, method int android.support.v7.widget.ListViewCompat.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
05-19 15:57:16.218 4343-4343/com.my.app E/Camera: Error 1
05-19 15:57:16.218 4343-4343/com.my.app W/System.err: java.lang.Exception
05-19 15:57:16.218 4343-4343/com.my.app W/System.err:     at io.fotoapparat.hardware.v1.Camera1.recordMethod(Camera1.java:304)
05-19 15:57:16.218 4343-4343/com.my.app W/System.err:     at io.fotoapparat.hardware.v1.Camera1.startPreview(Camera1.java:114)
05-19 15:57:16.218 4343-4343/com.my.app W/System.err:     at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:47)
05-19 15:57:16.218 4343-4343/com.my.app W/System.err:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
05-19 15:57:16.218 4343-4343/com.my.app W/System.err:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
05-19 15:57:16.218 4343-4343/com.my.app W/System.err:     at java.lang.Thread.run(Thread.java:761)
05-19 15:57:16.218 4343-4343/com.my.app D/AndroidRuntime: Shutting down VM
05-19 15:57:16.219 4343-4343/com.my.app E/AndroidRuntime: FATAL EXCEPTION: main
  Process: com.my.app, PID: 4343
  java.lang.IllegalStateException: Camera error code: 1
      at io.fotoapparat.hardware.v1.Camera1$1.onError(Camera1.java:73)
      at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1146)
      at android.os.Handler.dispatchMessage(Handler.java:102)
      at android.os.Looper.loop(Looper.java:154)
      at android.app.ActivityThread.main(ActivityThread.java:6121)
      at java.lang.reflect.Method.invoke(Native Method)
      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```
If I turn off filters, I also see this before the above:
```
05-19 16:03:22.249 671-23189/? I/CameraService: CameraService::connect call (PID -1 "com.my.app", camera ID 0) for HAL version default and Camera API version 1
05-19 16:03:22.261 679-10137/? D/NuPlayerDriver: notifyListener_l(0xe3899c00), (1, 0, 0), loop setting(0, 0)
05-19 16:03:22.274 679-10139/? D/NuPlayerDriver: notifyListener_l(0xe3899ae0), (1, 0, 0), loop setting(0, 0)
05-19 16:03:22.285 679-10141/? D/NuPlayerDriver: notifyListener_l(0xe3899120), (1, 0, 0), loop setting(0, 0)
05-19 16:03:22.285 671-23189/? I/Camera2ClientBase: Camera 0: Opened. Client: com.my.app (PID 10093, UID 10193)
05-19 16:03:22.287 10093-10135/com.my.app I/Adreno: QUALCOMM build                   : 6818200, Idb2b4cb785
                                                                Build Date                       : 11/17/16
                                                                OpenGL ES Shader Compiler Version: XE031.09.00.04
                                                                Local Branch                     : N25
                                                                Remote Branch                    : 
                                                                Remote Branch                    : 
                                                                Reconstruct Branch               : 
05-19 16:03:22.287 671-23189/? I/CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
05-19 16:03:22.289 1164-29056/? W/ActivityManager: Unable to start service Intent { cmp=com.google.android.apps.photos/.cameraassistant.CameraAssistantService (has extras) } U=0: not found
05-19 16:03:22.290 10093-10135/com.my.app I/OpenGLRenderer: Initialized EGL, version 1.4
05-19 16:03:22.290 10093-10135/com.my.app D/OpenGLRenderer: Swap behavior 1
[... cut ...]
05-19 16:03:23.984 671-10172/? E/QCamera: <HAL><ERROR> virtual int32_t qcamera::QCamera3RegularChannel::request(buffer_handle_t *, uint32_t, int &): 1967: On-the-fly buffer registration failed -12
05-19 16:03:23.984 671-10172/? E/QCamera: <HAL><ERROR> int qcamera::QCamera3HardwareInterface::processCaptureRequest(camera3_capture_request_t *): 4305: request failed
05-19 16:03:23.984 671-10172/? E/Camera3-Device: Camera 0: threadLoop: RequestThread: Unable to submit capture request 38 to HAL device: Out of memory (-12)
0
```

Diolor, 2017-05-19:
You are getting a [CAMERA_ERROR_UNKNOWN](https://developer.android.com/reference/android/hardware/Camera.html#CAMERA_ERROR_UNKNOWN). If android doesn't know what's the issue, I can't either :)

This is not a bug related to FA but the device's compatibility with camera1 API.

Which device are you using and more interestingly what API level? Alternatively you can force v2 as @dmitry-zaitsev suggested.

hagabaka, 2017-05-19:
I'm using Pixel XL, Android 7.1.2, and the app has `minSdkVersion 16`, `targetSdkVersion 25`. Forcing v2 worked for me, but will it be broken if the API level does not support v2?

hagabaka, 2017-05-19:
Actually, I got an error dialog when using the Android built-in camera app. So I rebooted the phone, and the problem has gone away without needing to force `cameraProvider`. So I guess this was just a temporary problem with my phone. Thanks for taking time to investigate!

closed 2017-05-19 by hagabaka