RedApparat/Fotoapparat #28: java.lang.IllegalArgumentException

FIndustries, 2017-05-23:
I get a lot of crashlogs on devices with Android 4.2 and 4.4 (most are Samsung):

`java.lang.IllegalArgumentException: 
  at io.fotoapparat.parameter.provider.InitialParametersValidator.validate(c.java:0)
                                                                  validateParameter
  at io.fotoapparat.parameter.provider.InitialParametersValidator.validate(c.java:0)
                                                                  validateParameter
  at io.fotoapparat.parameter.provider.InitialParametersProvider.initialParameters(b.java:0)
                                                                 putPreviewSize
                                                                 photoSize
  at io.fotoapparat.routine.StartCameraRoutine.run(d.java:0)
  at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
  at java.lang.Thread.run(Thread.java:841)`

I'm only using this code:
`fotoapparat = Fotoapparat
                    .with(this)
                    .into(cameraView)
                    .lensPosition(front())
                    .build();`

Diolor, 2017-05-23:
Hi @FIndustries ! Can you please put the full stacktrace and, if possible, non-obscufied?

FIndustries, 2017-05-24:
Unfortunately this is what I have.
It's from Google Play Developer Console.

dmitry-zaitsev, 2017-05-24:
I might have an idea what is the reason being the crash. Will fix it asap and publish a new version.

dmitry-zaitsev, 2017-05-24:
The issue should be fixed now. Try updating to `1.0.1`.

closed 2017-05-24 by dmitry-zaitsev