RedApparat/Fotoapparat #30: Crash

markGilchristBookingBug, 2017-05-25:
I am getting this crash via crashlytics

Fatal Exception: java.util.NoSuchElementException
       at java.util.HashMap$HashIterator.nextEntry(HashMap.java:789)
       at java.util.HashMap$KeyIterator.next(HashMap.java:814)
       at java.util.Collections.max(Collections.java:1626)
       at io.fotoapparat.parameter.selector.SizeSelectors$1.select(SizeSelectors.java:23)
       at io.fotoapparat.parameter.selector.SizeSelectors$1.select(SizeSelectors.java:20)
       at io.fotoapparat.parameter.selector.Selectors$2.select(Selectors.java:56)
       at io.fotoapparat.parameter.provider.InitialParametersProvider.putPreviewSize(InitialParametersProvider.java:67)
       at io.fotoapparat.parameter.provider.InitialParametersProvider.initialParameters(InitialParametersProvider.java:48)
       at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:41)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:818)

dmitry-zaitsev, 2017-05-25:
Which version of the library are you using?

markGilchristBookingBug, 2017-05-25:
'io.fotoapparat.fotoapparat:library:1.0.0', according to craslytics this has happened only on an htc device, all others work perfectly

dmitry-zaitsev, 2017-05-25:
This issue was just fixed yesterday and now available in 1.0.1. Please update :)

closed 2017-05-25 by dmitry-zaitsev

markGilchristBookingBug, 2017-05-25:
thanks guys

markGilchristBookingBug, 2017-05-25:
ATAL EXCEPTION: pool-4-thread-1
                                                                                      Process: com.groundcontrol.apps.android.maint, PID: 24826
                                                                                      java.util.NoSuchElementException
                                                                                          at java.util.HashMap$HashIterator.nextEntry(HashMap.java:789)
                                                                                          at java.util.HashMap$KeyIterator.next(HashMap.java:814)
                                                                                          at java.util.Collections.max(Collections.java:1630)
                                                                                          at io.fotoapparat.parameter.selector.SizeSelectors$1.select(SizeSelectors.java:23)
                                                                                          at io.fotoapparat.parameter.selector.SizeSelectors$1.select(SizeSelectors.java:20)
                                                                                          at io.fotoapparat.parameter.selector.Selectors$2.select(Selectors.java:56)
                                                                                          at io.fotoapparat.parameter.selector.Selectors$1.select(Selectors.java:26)
                                                                                          at io.fotoapparat.parameter.provider.InitialParametersProvider.putPreviewSize(InitialParametersProvider.java:69)
                                                                                          at io.fotoapparat.parameter.provider.InitialParametersProvider.initialParameters(InitialParametersProvider.java:50)
                                                                                          at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:41)
                                                                                          at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                                          at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                                          at java.lang.Thread.run(Thread.java:818)
we are still getting this on HTCs in 1.0.1