RedApparat/Fotoapparat #31: Crash version 1.0.0

markGilchristBookingBug, 2017-05-25:
Fatal Exception: java.lang.IllegalArgumentException: Opened camera does not support the selected focus_mode options.
       at io.fotoapparat.parameter.provider.InitialParametersValidator.validateParameter(InitialParametersValidator.java:31)
       at io.fotoapparat.parameter.provider.InitialParametersValidator.validate(InitialParametersValidator.java:24)
       at io.fotoapparat.parameter.provider.InitialParametersProvider.initialParameters(InitialParametersProvider.java:52)
       at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:41)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:841)

I am getting this on 1.0.0, if it now fixed then I am sorry to bother you

dmitry-zaitsev, 2017-05-25:
Please post here how you are creating Fotoapparat instance.

markGilchristBookingBug, 2017-05-25:
private void initCamera() {
        if (!checkCameraHardware(getContext()) || isInitialised) return;
        fotoapparat = Fotoapparat
                .with(getContext())
                .into(cameraView)
                .photoSize(biggestSize())
                .lensPosition(back())
                .focusMode( firstAvailable(  // (optional) use the first focus mode which is supported by device
                        continuousFocus(),
                        autoFocus(),         // in case if continuous focus is not available on device, auto focus will be used
                        fixed())             // if even auto focus is not available - fixed focus mode will be used
                )
                .build();
        isInitialised = true;
}

dmitry-zaitsev, 2017-06-07:
Fixed (hopefully) and should be rolled out in 1.0.2 within a day or two.

closed 2017-06-07 by dmitry-zaitsev