RedApparat/Fotoapparat #33: NPE version 1.0.0

markGilchristBookingBug, 2017-05-26:
Fatal Exception: java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()' on a null object reference
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:125)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:97)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.util.concurrent.ExecutionException: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()' on a null object reference
       at java.util.concurrent.FutureTask.report(FutureTask.java:94)
       at java.util.concurrent.FutureTask.get(FutureTask.java:164)
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:123)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:97)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()' on a null object reference
       at io.fotoapparat.result.transformer.BitmapPhotoTransformer.transform(BitmapPhotoTransformer.java:30)
       at io.fotoapparat.result.transformer.BitmapPhotoTransformer.transform(BitmapPhotoTransformer.java:13)
       at io.fotoapparat.result.PendingResult$1.call(PendingResult.java:55)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

dmitry-zaitsev, 2017-06-07:
Fixed. Fix will be available as part of 1.0.2.

Unfortunately it includes a small breaking change - `photoResult.whenAvailable()` now requires a `ErrorHandler` to be provided.

closed 2017-06-07 by dmitry-zaitsev

markGilchristBookingBug, 2017-06-11:
@dmitry-zaitsev  thanks for this I will upgrade, and run this on the office htc and hawae. I will post any more crashlytics things i see. thanks for this mark