RedApparat/Fotoapparat #36: onResult not called in android N

shayanpourvatan, 2017-06-08:
First of all thanks for your great library, I've ran sample code into android emulator API version 25, but `onResult` not called in `photoResult` for saved image, I've test your code in Sony API 23 and it's work fine.

Can you help me to fix this issue.

Diolor, 2017-06-08:
Thanks @shayanpourvatan for reporting. Might be affecting only emulator. N devices are ok for me but we need to check first. Might be related to file URIs/paths

Diolor, 2017-06-08:
Ok. Issue is that autofocus is not happening/completing in emulator

shayanpourvatan, 2017-06-08:
Thanks @Diolor

dmitry-zaitsev, 2017-06-11:
Implementation on emulator violates JavaDoc of the Camera, so, unfortunately, we are not able to help here without introducing hacks. If this issue will appear on a real device we will take a look again.

closed 2017-06-11 by dmitry-zaitsev