RedApparat/Fotoapparat #37: IllegalStateException - a.k - Critical exception reported by Firebug

jeffersonlicet, 2017-06-09:
```
 Exception java.lang.IllegalStateException: Preview stream is null. Make sure camera is opened.
io.fotoapparat.a.d.a.k ()
io.fotoapparat.a.d.a.h ()
io.fotoapparat.h.c.run ()
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1112)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:587)
java.lang.Thread.run (Thread.java:818)
```

The device : BLU STUDIO C 5+5 
Android OS: 5.0

Diolor, 2017-06-09:
Hi @jeffersonlicet  thanks for reporting this! Few questions:

1) How do you call FA's start/stop ?
2) Did you try reproduce to it? Would be very helpful if you provide us with the logs from our logger.

Thanks!

jeffersonlicet, 2017-06-09:
Thanks four your time.
I start/stop FA using a FotoapparatSwitcher located in this static OpenCV code:
```
private BaseLoaderCallback mLoaderCallback = new BaseLoaderCallback(this) {
        @Override
        public void onManagerConnected(int status) {
            switch (status) {
case LoaderCallbackInterface.SUCCESS: fotoapparatSwitcher.start();
break;}
        }
    };
```

OpenCV starts here:
```
@Override
    public void onResume() {
        super.onResume();

        try {
            if (!OpenCVLoader.initDebug()) {
                OpenCVLoader.initAsync(OpenCVLoader.OPENCV_VERSION_3_2_0, this, mLoaderCallback);
            } else mLoaderCallback.onManagerConnected(LoaderCallbackInterface.SUCCESS);
        } catch (Exception ex) {
            FirebaseCrash.report(ex);
        }
    }
```
I can't reproduce it, but i will try,  i think its related to the Blu phone.

dmitry-zaitsev, 2017-06-09:
Is `mLoaderCallback` always called from the main thread?

jeffersonlicet, 2017-06-09:
Yes, any clue?

dmitry-zaitsev, 2017-06-09:
Hm, not yet. If `mLoaderCallback` would be called from background thread that would explain the issue. I suspect that you can reproduce this issue by quickly switching between cameras while OpenCV is being initialized.

Would it be possible for you to provide us with logcat? Make sure you pass `logger` parameter while creating `Fotoapparat`.

dmitry-zaitsev, 2017-06-11:
I am closing this issue until there is more information available. As soon as we'll have logs I will reopen it.

closed 2017-06-11 by dmitry-zaitsev

jeffersonlicet, 2017-06-15:
Thank you very much. Solved starting the camera in the main thread as you told me. <3