RedApparat/Fotoapparat #39: Customize aperture

Diolor, 2017-06-11:
From #21 focusing only on camera aperture

makovkastar, 2018-01-25:
Is setting an aperture supported in Camera 1 API?

Diolor, 2018-01-26:
Honestly, I have not investigated this at all. According to [this SO post](https://stackoverflow.com/a/41041195/1447885) some phones support the `aperture` raw parameter, similar to how sensor sensitivity (iso) can be set. 

Have you tried something or do you have any information that would be helpful here? Feel free to share!

makovkastar, 2018-01-29:
This is the only way I have found so far too. Just wanted to ask you before starting the implementation of this feature.

makovkastar, 2018-01-29:
I have tested the following devices:

- Nexus 6
- Nexus 6P
- Samsung Galaxy S4
- Huawei P9 Lite
- Huawei Y360
- Sony Xperia Z3

None of them had the `aperture` or similar raw camera parameter. I'm not sure whether I can start working on this without having a phone that supports setting the aperture.

Diolor, 2018-02-02:
That's sad. An interesting idea I had in order to test this library across device specifics was to try firebase farm. From there we can maybe investigate many device's raw settings and see if any of supports custom aperture.

Still, I give low chance. Probably most of the phone cameras have a fixed aperture.

makovkastar, 2018-03-22:
Samsung Galaxy S9 and S9+ support variable aperture: https://www.cnet.com/how-to/how-samsung-galaxy-s9-variable-aperture-camera-works-mwc-2018/.