RedApparat/Fotoapparat #41: Wiki needs a bit of improvement

Reonarudo, 2017-06-13:
The wiki page is a bit badly structured making the Rx example very incomplete.
I would suggest adding a page with a better example of its usage, the needed/possible import versions like `compile 'io.fotoapparat.fotoapparat:adapter-rxjava2:1.0.2'` and maybe start adding Kotlin examples.

Diolor, 2017-06-13:
Thanks for the feedback @Reonarudo. What would you add regarding Kotlin?
Ofc PRs are welcome :)

Diolor, 2017-06-16:
The setup instructions are already under the relevant adapter README e.g. https://github.com/Fotoapparat/Fotoapparat/tree/master/fotoapparat-adapters/rxjava.

Should be easy to follow for a beginner in Rx. If not, please propose us enhancements :)

Honestly, there is much value in adding Kotlin examples since there are no huge differences between Java and Kotlin and one can easily understand both.

Will close for now but feel free to add more comments or open another issue if you need 👍

closed 2017-06-16 by Diolor