RedApparat/Fotoapparat #42: Rotate Screen camera zoomed

brian1212, 2017-06-15:
can you check it ,please.When i rotate screen portait to landscape, camera in landscape mode is zoomed,when i rotate back it normal.

dmitry-zaitsev, 2017-06-15:
Could you please provide screenshots?

brian1212, 2017-06-16:
My device set auto rotation.
when i open app in portrait and rotate to landscape.
[image]
[image]

when i open app in landscape mode and rotate to portrait
[image]
[image]

dmitry-zaitsev, 2017-06-16:
Thanks for the screenshots - looks odd indeed. Could you please also provide us with the following information:

- Which device is that? Which OS version?
- Snippet of code where you initialize Fotoapparat
- XML layout in which you added CameraView

brian1212, 2017-06-17:
i just run your demo.i test on nexus 4 rom NitrogenOS 7.1.2 and Samsung s5 rom 6.0.1.

dmitry-zaitsev, 2017-07-03:
@brian1212 May I ask you to do a little debugging for us? Could you please check what values `RendererParameters` has in `TextureRendererView` line 101 when you start the app and when you rotate the screen?

dmitry-zaitsev, 2017-07-03:
Alternatively, please try updating to 1.0.3. It will not fix the problem, but it will print more info to log, which would hopefully help us solve the issue.

namnguyen211, 2017-07-04:
when i start camera
 Renderer parameters are: RendererParameters{previewSize=Size{width=1280, height=960}, frameRotation=270}
when i rotate right
Renderer parameters are: RendererParameters{previewSize=Size{width=1280, height=960}, frameRotation=180}
when i rotate left
Renderer parameters are: RendererParameters{previewSize=Size{width=1280, height=960}, frameRotation=0}

seem to be when i rotate screen camera start again.right?

dmitry-zaitsev, 2017-07-07:
Hm, those seem to be correct values. Sorry to ask, but could you please also put a breakpoint at line 136 in `TexturerRendererView` and check which values `previewSize` take and which values are passed to this method call:

```java
getChildAt(0).layout(
                -extraX / 2,
                -extraY / 2,
                width - (extraX / 2),
                height - (extraY / 2)
);
```

brian1212, 2017-07-10:
when i open app:
[image]
when i rotate right
[image]
when i rotate left
[image]

Diolor, 2017-12-31:
Normally should not happen in v2. Feel free to comment

closed 2017-12-31 by Diolor