RedApparat/Fotoapparat #43: Issue About After Take a Picture

1060680253, 2017-06-19:
After taking pictures will be stuck for a while, this problem can be improved?But the system camera after taking pictures, will not be stuck for a while
Please help me out this problem。thankyou

Diolor, 2017-06-19:
Hi! Can you please give more concrete details? What stuck means? And also what phone/api versions did you experience it with?

1060680253, 2017-06-19:
[image]

you can see gif file，phone version is android 5.0.2

dmitry-zaitsev, 2017-07-03:
Sorry for slow reply. 

I looked into the issue and it seems that the root cause is that we are using Camera1 API by default since it is generally faster, less complicated and more reliable. You can switch to Camera2 API, which will fix that particular problem. 

Here is how you can do it. When you prepare your `Fotoapparat` instance, just provide `cameraProvider` parameter like that.

```
.cameraProvider(v2(context))
```

closed 2017-07-03 by dmitry-zaitsev