RedApparat/Fotoapparat #46: autofocus() callback

jeffersonlicet, 2017-06-26:
Hi, thank you very much for your time. is there any way to get the autofocus callback? seems like FocusResult is not available from getCurrentFotoapparat().autoFocus();

Thank you again.

dmitry-zaitsev, 2017-06-29:
Currently, the result of autofocus is not exposed, but we can improve that.

jeffersonlicet, 2017-06-30:
it will be awesome. Thank you.

dmitry-zaitsev, 2017-07-03:
Sorry for late reply, but could you please describe your use case a bit more? 

Currently, we are attempting to perform auto-focus for you when `takePicture` is called. So, I am evaluating the need for the actual result from `autoFocus`. On top of that, some devices are not always consistent with a result of the autofocus operation and might fail more often than you would normally expect them to, so handling autofocus result will get tricky pretty fast on a client side.

jeffersonlicet, 2017-07-03:
Using OpenCV to process a frame, but first it need to ensure the camera is focused , then I process the frame with a specific algorithm. Thank you. It also can be used performing face recognition (to draw rectangles etc)

dmitry-zaitsev, 2017-07-04:
OK, that makes sense. I will try to provide the solution until the end of this week.

If you are relying on frame processing you should also update to 1.0.4, which would noticeably improve the performance.

As for face detection - we just started working on a module for Fotoapparat which would do face detection without relying on Google Play Services: https://github.com/Fotoapparat/FaceDetector

dmitry-zaitsev, 2017-07-08:
The feature is in master. You can expect it in the new release which will happen until the end of the week.

closed 2017-07-08 by dmitry-zaitsev

jeffersonlicet, 2017-07-09:
Thank you very much.

omarDG, 2020-10-22:
Hi,

Where to find the documentation related to the focus callback? I'm unable to find it.

Regards