RedApparat/Fotoapparat #48: setParamaters Crash on Haiwai,1.0.2

markGilchristBookingBug, 2017-06-28:
Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:2028)
       at io.fotoapparat.hardware.v1.Camera1.updateParameters(Camera1.java:178)
       at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:40)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573)
       at java.lang.Thread.run(Thread.java:838)

dmitry-zaitsev, 2017-06-29:
Could you please provide us with more logs from the device? In particular, I am interested in logs with tag `Fotoapparat`

markGilchristBookingBug, 2017-06-29:
[image]

dmitry-zaitsev, 2017-06-29:
When creating `Fotoapparat` instance, you can specify the `Logger`. In this logger, you can try to write more information to Crashlytics.

dmitry-zaitsev, 2017-07-03:
Few more generic questions which should help us to analyze the problem:
- Is app always crashing on this model? Or were there some successful sessions?
- Do you have more related crashes on the same type of the device? Or is it just this one user?

dmitry-zaitsev, 2017-07-03:
@markGilchristBookingBug In the meanwhile, please try updating to 1.0.3. It will not fix the problem itself, but it will include more information into exception's message.

arekolek, 2017-07-06:
I have the same issue (exactly the same stack trace). I've added Crashlytics logging and upgraded to ver. 1.0.3 of Fotoapparat, maybe tomorrow I'll have something interesting to share.

Crashlytics reports that 11 users on 6 different device models are affected in my case. The phone models being:

- CHC-U01
- GT-I9060I
- GT-I9195
- GT-S7580
- SM-G355HN
- SM-J100H

At the moment, I don't have a way of saying whether any of those models had a successful session.

Because there are more users than models, I suspect that there must be multiple users affected by this issue on some device models.

Diolor, 2017-07-06:
Hi @markGilchristBookingBug  + @arekolek, 

How do you set the preview sizes? Potentially it's the same issue with [this SO question](https://stackoverflow.com/a/3964460/1447885).

arekolek, 2017-07-07:
@Diolor I just have the `CameraView` in my layout:

```xml
<io.fotoapparat.view.CameraView
        android:id="@+id/camera_preview"
        android:layout_above="@+id/bottom_panel"
        android:layout_width="match_parent"
        android:layout_height="match_parent"/>
```

and set it up like this in my fragment:

```java
@BindView(R.id.camera_preview) CameraView cameraPreview;
private Fotoapparat fotoapparat;

@Override
public void onViewCreated(View view, @Nullable Bundle savedInstanceState) {
    ButterKnife.bind(this, view);
    fotoapparat = Fotoapparat.with(getActivity())
            .into(cameraPreview)
            .photoSize(biggestSize())
            .lensPosition(front())
            .focusMode(firstAvailable(
                    continuousFocus(),
                    autoFocus(),
                    fixed()
            ))
            .logger(new Logger() {
                @Override
                public void log(String msg) {
                    Crashlytics.log(Log.DEBUG, "Fotoapparat", msg);
                }
            })
            .build();
}

@Override
public void onStart() {
    super.onStart();
    fotoapparat.start();
}
```

I'm not doing anything with `Camera.Parameters` directly in my code, so it's whatever Fotoapparat does by default.

dmitry-zaitsev, 2017-07-07:
It seems (obviously) that Camera is not happy with whichever parameters we are passing to it. At the moment we try to update the whole set of parameters at once, so it's hard to tell which property causes the issue exactly.

I'll create a fallback mechanism which will try to update each parameter individually and log/suppress the failures. Expect the new versions until the end of the week.

Another topic. I see you are tracking Fotoapparat logs in Crashlytics. Is there any chance you can share those with us?

dmitry-zaitsev, 2017-07-08:
The experimental fix is ready. You can try it out now in new version `1.1.0`.

arekolek, 2017-07-10:
@dmitry-zaitsev I've added Crashlytics only recently, but got some reports over the weekend. Looks like they got obfuscated by ProGuard, but there were only at most three lines there anyway, so I guess those must be:

> 07-10 12:10:41.700  1755  2487 D Fotoapparat: open
> 07-10 12:10:41.862  1755  2487 D Fotoapparat: getCapabilities
> 07-10 12:10:41.868  1755  2487 D Fotoapparat: updateParameters

Sometimes all three, sometimes only two first, other times only the first, and sometimes none. Additionally, the exception message is more informative now and says:

```
io.fotoapparat.hardware.CameraException: Failed to set parameters:
       Parameters{values={FLASH=OFF, FOCUS_MODE=FIXED,
              PREVIEW_SIZE=Size{width=1280, height=720},
              PICTURE_SIZE=Size{width=1920, height=1080}}}
       at io.fotoapparat.hardware.v1.Camera1.throwOnFailSetParameters(SourceFile:59)
       at io.fotoapparat.hardware.v1.Camera1.updateParameters(SourceFile:196)
       at io.fotoapparat.routine.StartCameraRoutine.run(SourceFile:40)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:841)
Caused by java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:1753)
       at io.fotoapparat.hardware.v1.Camera1.updateParameters(SourceFile:194)
       at io.fotoapparat.routine.StartCameraRoutine.run(SourceFile:40)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:841)
```

That's on Samsung Galaxy S4 mini (GT-I9195). On ASUS MeMO Pad HD 7 (ME173X) the parameters were: `{FLASH=OFF, FOCUS_MODE=FIXED, PREVIEW_SIZE=Size{width=1280, height=960}, PICTURE_SIZE=Size{width=1280, height=960}}`

Which kind of makes sense, considering S4 mini has a 540x960 px display and ASUS has a 800x1280 px display (from what I could find), so the requested preview size is bigger than the display in both instances. But it baffles me why would `Camera.Parameters.getSupportedPreviewSizes()` return such sizes in that case.

I'll try to keep ProGuard from obfuscating the logs and update Fotoapparat to 1.1.0 and we'll see what happens.

dmitry-zaitsev, 2017-07-11:
Wow, that's really helpful!

> Which kind of makes sense, considering S4 mini has a 540x960 px display and ASUS has a 800x1280 px display (from what I could find), so the requested preview size is bigger than the display in both instances

That is actually OK, preview stream might have higher resolution than the screen itself.

I am very suspicious about `FOCUS_MODE=FIXED` - that is a rather unusual parameter and we use it by default if no other focus mode is specified. I assumed that it is supported by all devices, but apparently that is not the case. I will try to fix that.

Either way, latest release (`1.1.0`) will fix that issue since invalid parameters will just be suppressed. Not ideal from code perspective, but will do until we'll roll out the proper fix.

dmitry-zaitsev, 2017-08-26:
The way it is implemented right now seems to be quite robust, so I am closing the issue.

closed 2017-08-26 by dmitry-zaitsev