RedApparat/Fotoapparat #49: question how to save photo as byte array

yp25, 2017-06-29:
Hi, I want to know how to save photo captured as byte array directly. 
In this sample code : 
`private void takePicture() {
        PhotoResult photoResult = fotoapparatSwitcher.getCurrentFotoapparat().takePicture();

        photoResult.saveToFile(new File(
                getExternalFilesDir("photos"),
                "photo.jpg"
        ));

        photoResult
                .toBitmap(scaled(0.25f))
                .whenAvailable(new PendingResult.Callback<BitmapPhoto>() {
                    @Override
                    public void onResult(BitmapPhoto result) {
                        ImageView imageView = (ImageView) findViewById(R.id.result);

                        imageView.setImageBitmap(result.bitmap);
                        imageView.setRotation(-result.rotationDegrees);
                    }
                });
    }

`

I can read from photo.jpg and then save to byte array variable . But how to directly read the photo data as byte array? Thank you

dmitry-zaitsev, 2017-06-29:
`PhotoResult` actually delivers photo as a byte array, and then `toBitmap` converts it to actual `Bitmap`. So, what you can do:

```java
photoResult
    .toPendingResult()
    .whenAvailable(photo -> {
        // photo.encodedImage - that's your byte array
        // pay attention to photo.rotationDegrees parameter
    });
```

closed 2017-06-29 by dmitry-zaitsev

ElyDantas, 2020-03-08:
> `PhotoResult` actually delivers photo as a byte array, and then `toBitmap` converts it to actual `Bitmap`. So, what you can do:
>
> ```java
> photoResult
>     .toPendingResult()
>     .whenAvailable(photo [...]

Should be in the README :)