RedApparat/Fotoapparat #50: Fatal Exception: io.fotoapparat.hardware.CameraException java.lang.RuntimeException: Fail to connect to camera service

HaroldHibari, 2017-07-03:
Hi
One of our users is getting these crashes whilst trying to use the camera. 
Phone model: HTC One mini 2
Android version: 4.4.2

**Stack trace 1:**
Fatal Exception: io.fotoapparat.hardware.CameraException: java.lang.RuntimeException: Fail to connect to camera service
       at io.fotoapparat.hardware.v1.Camera1.open(Camera1.java:63)
       at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:39)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:864)
Caused by java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.native_setup(Camera.java)
       at android.hardware.Camera.<init>(Camera.java:549)
       at android.hardware.Camera.open(Camera.java:424)
       at io.fotoapparat.hardware.v1.Camera1.open(Camera1.java:60)
       at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:39)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:864)

**Stack trace 2:**
Fatal Exception: java.lang.NullPointerException
       at io.fotoapparat.hardware.v1.Camera1.stopPreview(Camera1.java:123)
       at io.fotoapparat.routine.StopCameraRoutine.run(StopCameraRoutine.java:18)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:864)

dmitry-zaitsev, 2017-07-03:
Hi, thanks for the report!

Few generic questions which should help us to analyze the problem:
- Is issue reproducible? If yes, how easy is it to reproduce?
- If app just crashes in 100% of the cases for this device, do other camera apps work? Sometimes it can be that camera hardware stops working in all apps. Restarting the device should help.
- Do you have more related crashes on the same type of the device? Or is it just this one user?

dmitry-zaitsev, 2017-07-03:
In the meanwhile, please try updating to 1.0.3. It will not fix the problem itself, but it will include more information into exception's message.

HaroldHibari, 2017-07-04:
###    **1.   Is issue reproducible? If yes, how easy is it to reproduce?**
Completely reproducible. Happens every time you try to use the camera on said phone.

###  **2.   If app just crashes in 100% of the cases for this device, do other camera apps work?** Sometimes it can be that camera hardware stops working in all apps. Restarting the device should help.
Yes. It was working previously with another camera library (https://github.com/gogopop/CameraKit-Android) which we decided to swap out to fix intermittent crashes.

###  **3.   Do you have more related crashes on the same type of the device? Or is it just this one user?**

Yes. It has also crashed on a Samsung SM-G900F as well as Huawei Ascend Y330. A new stack trace for version 1.0.2

Fatal Exception: java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: startPreview failed
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: startPreview failed
       at java.util.concurrent.FutureTask.report(FutureTask.java:94)
       at java.util.concurrent.FutureTask.get(FutureTask.java:164)
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:127)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.util.concurrent.ExecutionException: java.lang.RuntimeException: startPreview failed
       at java.util.concurrent.FutureTask.report(FutureTask.java:94)
       at java.util.concurrent.FutureTask.get(FutureTask.java:164)
       at io.fotoapparat.result.PendingResult$1.call(PendingResult.java:56)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at io.fotoapparat.hardware.v1.Camera1.startPreview(Camera1.java:116)
       at io.fotoapparat.routine.TakePictureTask$1.call(TakePictureTask.java:34)
       at io.fotoapparat.routine.TakePictureTask$1.call(TakePictureTask.java:16)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

arekolek, 2017-07-07:
It's rather easy to reproduce this on any device. Just turn on the flashlight, then go to your app that uses Fotoapparat. I doubt that was what happened in those other instances, but documentation of [`Camera.open(int)`](https://developer.android.com/reference/android/hardware/Camera.html#open(int)) tells you that it throws a:

> RuntimeException - if opening the camera fails (for example, if the camera is in use by another process or device policy manager has disabled the camera).

I don't see how I can prevent my app from crashing in that scenario, as a consumer of the Fotoapparat API.

dmitry-zaitsev, 2017-07-07:
@arekolek thanks for the suggestion. I can't really reproduce it on Samsung S7 Edge, but I will try to find other devices.

Either way, if what you describe is true - we can't really provide an access to camera, but we can at least allow you to handle the error yourself.

dmitry-zaitsev, 2017-07-12:
I am closing it in favor of #55 

From what I can tell now - a camera just can't be opened because the system does not allow us to do so. We pass exactly 1 parameter, so there is not much we can do wrong. When there will be more information available, we'll reopen it.

closed 2017-07-12 by dmitry-zaitsev