RedApparat/Fotoapparat #51: FrameProcessor - Byte Array to Bitmap

bsobe, 2017-07-04:
I try to convert frame.image(Byte Array) to Bitmap with this code
`BitmapFactory.decodeByteArray(frame.image, 0, frame.image.length);`

But, bitmap always return null. How can I convert frame.image to bitmap? 

**Logcat message: --- SkImageDecoder::Factory returned null**

dmitry-zaitsev, 2017-07-05:
Frame images are in NV21 format and BitmapFactory is not able to decode them and returns null.

There are various solutions available. Here is one of them: https://stackoverflow.com/a/32793861/926907

Out of curiousity, what are you going to do with those images? Maybe I can provide some advice.

closed 2017-07-05 by dmitry-zaitsev

bsobe, 2017-07-05:
I am developing real time OCR application with Tesseract (tess-two).
Also I need to crop preview (to improve success rate of OCR) with Rect like Barcode libs (barcode libs show full screen preview but only scan specified area). 

Can you give a suggestion?