RedApparat/Fotoapparat #52: takePicture never called

ananduri, 2017-07-06:
Hey guys, thanks for this very useful library.

__Problem__:
When using a Moto G4, I can take a picture once. But the second time I try to take a picture, it doesn't work. 

__Details__:
* When it fails, I see autoFocus in the Fotoapparat logs, but takePicture never shows up.
* The OS version was api 25
* I tried this on 4 phones:
  - Samsung S5: no issue
  - Samsung S6: no issue
  - Moto G: no issue
  - Moto G4: fails
* Snippet where I initialize fotoapparat:
```
fotoapparat = Fotoapparat
                .with(this)
                .into(cameraView)
                .lensPosition(back())
                .logger(loggers(logcat()))
                .focusMode(firstAvailable(  
                       continuousFocus(),
                       autoFocus(),        
                       fixed()             
                ))
                .photoSize(wideRatio(biggestSize()))
                .build();
```
* If I change the snippet to:
```
fotoapparat = Fotoapparat
                .with(this)
                .into(cameraView)
                .lensPosition(back())
                .logger(loggers(logcat()))
                .focusMode(autoFocus()) //  <---  this line changed
                .photoSize(wideRatio(biggestSize()))
                .build();
```
this issue doesn't arise. But then obviously I can't use continuous focus mode.

Unfortunately, I don't have access to the problematic device right now, so I can't do additional testing. If this issue isn't solved because of that, feel free to close it, and I can open an issue again when I get access to the device.

dmitry-zaitsev, 2017-07-07:
Thanks for the report!

We encountered somewhat similar (although not exactly the same) problem on an an emulator, but this is the first time we hear about it happening on a real device. Will try to fix it 👍

dmitry-zaitsev, 2017-07-07:
OK, that was an easy fix. You can expect it in the next version (should be available until the end of the week).

closed 2017-07-07 by dmitry-zaitsev