RedApparat/Fotoapparat #56: pausing app while taking picture crashes

ananduri, 2017-07-12:
**Problem:**
When I hit the home, back, or recents buttons right after taking a picture, the app crashes.

**Details:**
I have my code setup as follows:
```
    public void takePhotoOnClick() {
        PhotoResult photoResult = fotoapparat.takePicture();

        photoResult.toPendingResult().whenAvailable(new PendingResult.Callback<Photo>() {
            @Override
            public void onResult(Photo photo) {
                //do stuff
            }
        });
    }
```
This function is called when I click a button. If I pause the app right after hitting that button, I get a crash like this:
```
D/Fotoapparat: open
D/Fotoapparat: getCapabilities
D/Fotoapparat: updateParameters
D/Fotoapparat: setDisplayOrientation
D/Fotoapparat: getRendererParameters
D/Fotoapparat: Renderer parameters are: 80}, frameRotation=270}
D/Fotoapparat: setDisplaySurface
D/Fotoapparat: startPreview
D/Fotoapparat: autoFocus
D/Fotoapparat: takePicture
D/Fotoapparat: startPreview
E/BufferQueue: [unnamed-15408-2] query: BufferQueue has been abandoned!
D/Fotoapparat: stopPreview
E/AndroidRuntime: FATAL EXCEPTION: pool-4-thread-1
                                                                         Process: com.appname.appname, PID: 15408
                                                                         java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: startPreview failed
                                                                             at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
                                                                             at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
                                                                             at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
                                                                             at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                                             at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                                             at java.lang.Thread.run(Thread.java:811)
                                                                          Caused by: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: startPreview failed
                                                                             at java.util.concurrent.FutureTask.report(FutureTask.java:93)
                                                                             at java.util.concurrent.FutureTask.get(FutureTask.java:163)
                                                                             at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:127)
                                                                             at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20) 
                                                                             at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98) 
                                                                             at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112) 
                                                                             at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587) 
                                                                             at java.lang.Thread.run(Thread.java:811) 
                                                                          Caused by: java.util.concurrent.ExecutionException: java.lang.RuntimeException: startPreview failed
                                                                             at java.util.concurrent.FutureTask.report(FutureTask.java:93)
                                                                             at java.util.concurrent.FutureTask.get(FutureTask.java:163)
                                                                             at io.fotoapparat.result.PendingResult$1.call(PendingResult.java:56)
                                                                             at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                                             at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112) 
                                                                             at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587) 
                                                                             at java.lang.Thread.run(Thread.java:811) 
                                                                          Caused by: java.lang.RuntimeException: startPreview failed
                                                                             at android.hardware.Camera.startPreview(Native Method)
                                                                             at io.fotoapparat.hardware.v1.Camera1.startPreview(Camera1.java:135)
                                                                             at io.fotoapparat.routine.picture.TakePictureTask$1.call(TakePictureTask.java:34)
                                                                             at io.fotoapparat.routine.picture.TakePictureTask$1.call(TakePictureTask.java:16)
                                                                             at java.util.concurrent.FutureTask.run(FutureTask.java:237) 
                                                                             at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112) 
                                                                             at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587) 
                                                                             at java.lang.Thread.run(Thread.java:811) 
07-12 13:26:17.412 15408-15603/com.appname.appname D/Fotoapparat: close
```
However, if I don't call `.whenAvailable()` on `photoResult.toPendingResult()`, no crash occurs. These results are true for both my devices (which are Samsung S5 and Moto G).

Maybe you guys have seen such a problem before? Is there an easy way to prevent this that I'm overlooking?

dmitry-zaitsev, 2017-07-12:
Thanks for the reporting. It will be fixed soon.

dmitry-zaitsev, 2017-07-14:
Done. Expect the fix in the next release which would happen until the end of this week.

It is not reproducible on all devices. Some devices are not happy when their surface view is detached from the window. We have no way of knowing that, so we are ignoring such exception. There is no harm in doing so since preview will be restarted after Activity will be resumed.

closed 2017-07-14 by dmitry-zaitsev