RedApparat/Fotoapparat #57: Permissions were not added to manifest

scifinder, 2017-07-14:
Hello! I'm new to Android Studio. By the instruction I connected your library, but in the manifest file no permissions were added. Why?

bachhuberdesign, 2017-07-14:
You have to manually add permissions to manifest. You will also need to handle the runtime permissions in your code. This is not a Fotoapparat issue; it is the same with every library on Android.

dmitry-zaitsev, 2017-07-14:
Permissions will be added automatically when you try to build the app. You will not see them in your AndroidManifest, as they are added by the manifest merger.

If you are interested about the details, you can read about it here: https://developer.android.com/studio/build/manifest-merge.html

closed 2017-07-14 by dmitry-zaitsev