RedApparat/Fotoapparat #60: Manual Focus

Jameido, 2017-07-17:
Hi there, 
it's possible to let the user choose which part of the picture should be focused manually? 
Like on the default android camera, when the use taps on the screen it should set the focus on that area, as shown in the attached gif

[image]

dmitry-zaitsev, 2017-07-19:
Are you looking for a way to focus on a specific point on a picture or just tap-to-focus feature? The former is not yet implemented, the latter is already available.

Jameido, 2017-07-19:
I'm looking for the first one, it would be a nice add to this amazing library ✨

TemMax, 2017-07-27:
+1. Add please feature for focus on specific point and pinch-to-zoom feature.
Thanks for wonderful library!

dmitry-zaitsev, 2017-08-26:
Sorry guys, this feature will not make it into 1.3.0 as it is a bit tricky to implement. We will implement it as part of 1.4.0.

Also, if someone is eager to help - please contact me. I can provide the directions on how to implement this feature in the library.

friendoye, 2017-09-02:
@dmitry-zaitsev I would like to step in, if you don't mind.

dmitry-zaitsev, 2017-09-04:
@friendoye that would be great!

First of all, let's forget about Camera2 for now. We do not actively support it anymore (since it is surprisingly more buggy than Camera1), so let's focus on Camera1.

To focus on the specific area we can use `Camera.Parameters.setFocusAreas`. There are some caveats though:

- We should first call `getMaxNumFocusAreas()` to check if focus areas are supported at all.
- We need to map user touches to `Area` class. But camera preview stream might actually be bigger than the size of the actual view. There is a handy layout which we use for something similar in our `FaceDetector` library - [CameraOverlayLayout](https://github.com/Fotoapparat/FaceDetector/blob/master/Android/facedetector/src/main/java/io/fotoapparat/facedetector/view/CameraOverlayLayout.java) We might just copy it into `Fotoapparat`. I believe people who work with Computer Vision will find it very handy.
- Area coordinates are in camera sensor space, which means that they are different from what user sees on his screen. That means that we have to rotate the area (and possibly even mirror it) before calling `setFocusAreas`. `OrientationUtils` might come in handy here.

friendoye, 2017-09-04:
@dmitry-zaitsev hmm, sounds interesting! I'll start working on PR this week.

abodehBabelli, 2017-09-13:
@friendoye is there any estimated date to finish adding manual focus??

friendoye, 2017-09-13:
@abodehBabelli I'm planning to submit PR at the end of this week. It will include tap to focus feature for Camera1 API.

Diolor, 2018-01-08:
@ all. Tap to focus feature has been included in v2.1 🎉

closed 2018-01-08 by Diolor

alexischenko, 2018-04-24:
Can someone help? I can't find focusView method in Fotoapparat builder class. I'm using version 2.2.0

Diolor, 2018-04-24:
Did you check the sample?

Avalin, 2018-05-29:
I copied the sample to my project, and the focusView is for some reason not being recognised on mine either. Rest works.

chungyau97, 2019-05-30:
@Diolor @friendoye @Avalin 
How to achieve fix value manual focus instead of using tap assisted focus?
#374 (comment)