RedApparat/Fotoapparat #64: photoResult not full view in imageview

faketon, 2017-07-26:
photoResult not full view in imageview what i miss ?

i try to setScaleType in imageview but not work.
can you explane me what i miss ?

faketon, 2017-07-26:
But i use Glide to help load image from path file. it can full view.
But i want to know this lib can help or not. Thank you for lib

Diolor, 2017-07-26:
This library has nothing to do with loading images to an imageview. Please see https://developer.android.com/reference/android/widget/ImageView.ScaleType.html

closed 2017-07-26 by Diolor

bruferrari, 2017-07-31:
I also have this problem and even using scaleType it does not fit the entire screen, I also had changed the sample and can't achieve the expected results.

dmitry-zaitsev, 2017-08-01:
@faketon @bruferrari could you please provide screenshots for the problem? It would be easier to understand what the problem is.

bruferrari, 2017-08-01:
Hi @dmitry-zaitsev here are some pics to better describe the mentioned problem, note that with any kind of scaleType that I'd used the image does not goes beyond that limit and I have the CameraView with width and height in match_parent.

Please do not pay so much attention on that odd floating FAB, I was trying everything to solve the problem even swap between ConstraintLayout to FrameLayout in order to note some changes but without success.

[image]
`scaleType="fitXY"`

[image]
`original photo saved`

[image]
`without scale type`

faketon, 2017-08-02:
i fix my problem by use glide to load image from path. if you use bitmap from photoResult to set in imageview that photo can't full view. @bruferrari that is my resovled

dmitry-zaitsev, 2017-08-02:
Note that in sample app we are using `scaled ()` when taking a photo. Removing that should solve your problem.