RedApparat/Fotoapparat #66: takePicture() crash

GiovanniM, 2017-07-26:
Exception java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
io.fotoapparat.result.PendingResult.fromFuture (PendingResult.java)
<OR>.transform (PendingResult.java)
<OR>.whenAvailable (PendingResult.java)
<OR>.notifyCallbackOnMainThread (PendingResult.java)
<OR>.getResultUnsafe (PendingResult.java)
<OR>.access$000 (PendingResult.java)
<OR>.access$200 (PendingResult.java)
io.fotoapparat.result.PendingResult.whenDone (PendingResult.java)
<OR>.access$100 (PendingResult.java)
io.fotoapparat.result.PendingResult$2.run (PendingResult.java)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1113)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:588)
java.lang.Thread.run (Thread.java:818)
arrow_drop_down
Caused by java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
java.util.concurrent.FutureTask.report (FutureTask.java:94)
java.util.concurrent.FutureTask.get (FutureTask.java:164)
io.fotoapparat.result.PendingResult.fromFuture (PendingResult.java)
<OR>.transform (PendingResult.java)
<OR>.whenAvailable (PendingResult.java)
<OR>.notifyCallbackOnMainThread (PendingResult.java)
<OR>.getResultUnsafe (PendingResult.java)
<OR>.access$000 (PendingResult.java)
<OR>.access$200 (PendingResult.java)
io.fotoapparat.result.PendingResult.whenDone (PendingResult.java)
<OR>.access$100 (PendingResult.java)
io.fotoapparat.result.PendingResult$2.run (PendingResult.java)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1113)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:588)
java.lang.Thread.run (Thread.java:818)
arrow_drop_down
Caused by java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
java.util.concurrent.FutureTask.report (FutureTask.java:94)
java.util.concurrent.FutureTask.get (FutureTask.java:164)
io.fotoapparat.result.PendingResult$1.call (PendingResult.java)
java.util.concurrent.FutureTask.run (FutureTask.java:237)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1113)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:588)
java.lang.Thread.run (Thread.java:818)
arrow_drop_down
Caused by java.lang.RuntimeException: takePicture failed
android.hardware.Camera.native_takePicture (Camera.java)
android.hardware.Camera.takePicture (Camera.java:1492)
io.fotoapparat.hardware.v1.Camera1.takePicture (Camera1.java)
io.fotoapparat.routine.picture.TakePictureTask$1.call (TakePictureTask.java)
io.fotoapparat.routine.picture.TakePictureTask$1.call (TakePictureTask.java)
java.util.concurrent.FutureTask.run (FutureTask.java:237)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1113)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:588)
java.lang.Thread.run (Thread.java:818)

User
Country Code: —
Carrier Code: —
Carrier: I WIND
Locale: it-IT

Performance
VM free: 10.01MB
VM total: 75.43MB
VM max: 192MB
Battery level: 58%
Charging state: Unplugged
Connection State: Wwan

Device
Manufacturer: Xiaomi
Model: Redmi 4
Board: Msm8953
Android API: 23
Android OS: 6.0.1
Brand: Xiaomi
RAM: 2.78GB
Orientation: Portrait
Proximity to user: —

Error
Date: Jul 25, 2017, 7:15:53 PM
App Version: 10005 (1.0.5)

GiovanniM, 2017-07-27:
Sorry, when you see only .METHOD_NAME instead of the full class path there is an OR, firebase gave me the stacktrace as I pasted

GiovanniM, 2017-07-27:
An additional stack trace states error=-38

dmitry-zaitsev, 2017-07-27:
Could you please show us the code which you use to take picture?

dmitry-zaitsev, 2017-07-27:
Looks like this issue is a duplicate of #65 , so I am closing this one in favor of another one.

closed 2017-07-27 by dmitry-zaitsev