RedApparat/Fotoapparat #68: tryToStartCamera() crash

GiovanniM, 2017-07-27:
Exception java.lang.NullPointerException:
io.fotoapparat.hardware.v1.Camera1.throwOnFailSetDisplaySurface (Camera1.java)
<OR>.open (Camera1.java)
<OR>.throwOnFailedToOpenCamera (Camera1.java)
<OR>.close (Camera1.java)
<OR>.throwOnFailStartPreview (Camera1.java)
<OR>.setDisplaySurface (Camera1.java)
<OR>.setDisplayOrientation (Camera1.java)
<OR>.computeDisplayOrientation (Camera1.java)
<OR>.updateParameters (Camera1.java)
<OR>.logRendererParameters (Camera1.java)
<OR>.logFailedAutoFocus (Camera1.java)
<OR>.access$000 (Camera1.java)
io.fotoapparat.routine.StartCameraRoutine.tryToStartCamera (StartCameraRoutine.java)
io.fotoapparat.routine.StartCameraRoutine.run (StartCameraRoutine.java)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1112)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:587)
java.lang.Thread.run (Thread.java:841)

User
Country Code: —
Carrier Code: —
Carrier: vodafone IT
Locale: it-CH

Performance
VM free: 1.43MB
VM total: 10.64MB
VM max: 128MB
Battery level: 1%
Charging state: Unplugged
Connection State: Wwan

Device
Manufacturer: XTOUCH
Model: X4shot
Board: Xtouch
Android API: 19
Android OS: 4.4.2
Brand: Xtouch
RAM: 966.19MB
Orientation: Portrait
Proximity to user: —

Error
Date: Jul 26, 2017, 5:32:46 PM
App Version: 10005 (1.0.5)

GiovanniM, 2017-07-27:
Sorry, when you see only .METHOD_NAME instead of the full class path there is an OR, firebase gave me the stacktrace as I pasted

dmitry-zaitsev, 2017-07-27:
Could you please provide your XML layout and how you create Fotoapparat instance?

GiovanniM, 2017-07-28:
Layout

<?xml version="1.0" encoding="utf-8"?>
<FrameLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    android:orientation="vertical"
    android:layout_width="match_parent"
    android:layout_height="match_parent">

    <io.fotoapparat.view.CameraView
        android:id="@+id/camera_view"
        android:layout_width="match_parent"
        android:layout_height="match_parent"/>

    <android.support.design.widget.FloatingActionButton
        android:id="@+id/take_photo_button"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="end|bottom"
        android:layout_margin="8dp"
        android:text="@string/next"
        app:fabSize="normal"
        app:srcCompat="@drawable/ic_done" />

</FrameLayout>

and this method is called in the onCreate()

private void initFotoApparat() {
        fotoapparat = Fotoapparat
                .with(this)
                .into(cameraView)
                .lensPosition(Selectors.single(LensPosition.BACK))
                .cameraErrorCallback(cameraErrorCallback)
                .logger(Loggers.logcat())
                .build();
    }

The crash has been seen only 1 time. The app is used by almost 100 persons from less than a week

dmitry-zaitsev, 2017-08-01:
Closing since there is not enough information at the moment and it looks like a device specific issue. If this issue will be reproducible on more than 1 device I will reopen it.

closed 2017-08-01 by dmitry-zaitsev