RedApparat/Fotoapparat #69: 4 Uncaught Fatal Exceptions in TakePictureTask & GetCapabilitiesTask on Galaxy Nexus

jpribble, 2017-07-27:
Four fatal, uncaught exceptions were found while testing on the Galaxy Nexus. These crashes are occurring in background threads that the caller cannot catch. Can you please help catch these exceptions and return them to the caller via the cameraErrorCallback? Thank you.

**1) Galaxy Nexus running 4.2.2 - Crash while taking picture:**
07-21 14:55:24.170 3552-3580/? E/AndroidRuntime: FATAL EXCEPTION: pool-3-thread-1
                                                 Process: com.example.gbs131.fotoapparat, PID: 3552
                                                 java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
                                                     at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
                                                     at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
                                                     at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
                                                     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                     at java.lang.Thread.run(Thread.java:818)
                                                  Caused by: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
                                                     at java.util.concurrent.FutureTask.report(FutureTask.java:93)
                                                     at java.util.concurrent.FutureTask.get(FutureTask.java:163)
                                                     at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:127)
                                                     at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20) 
                                                     at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98) 
                                                     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112) 
                                                     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587) 
                                                     at java.lang.Thread.run(Thread.java:818) 
                                                  Caused by: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
                                                     at java.util.concurrent.FutureTask.report(FutureTask.java:93)
                                                     at java.util.concurrent.FutureTask.get(FutureTask.java:163)
                                                     at io.fotoapparat.result.PendingResult$1.call(PendingResult.java:56)
                                                     at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112) 
                                                     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587) 
                                                     at java.lang.Thread.run(Thread.java:818) 
                                                  Caused by: java.lang.RuntimeException: takePicture failed
                                                     at android.hardware.Camera.native_takePicture(Native Method)
                                                     at android.hardware.Camera.takePicture(Camera.java:1436)
                                                     at io.fotoapparat.hardware.v1.Camera1.takePicture(Camera1.java:258)
                                                     at io.fotoapparat.routine.picture.TakePictureTask$1.call(TakePictureTask.java:24)
                                                     at io.fotoapparat.routine.picture.TakePictureTask$1.call(TakePictureTask.java:19)
                                                     at java.util.concurrent.FutureTask.run(FutureTask.java:237) 
                                                     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112) 
                                                     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587) 
                                                     at java.lang.Thread.run(Thread.java:818) 

**2) Galaxy Nexus running 4.2.2 - Crash when first launching MainActivity:**
07-25 14:17:37.097 13588-13617/com.example.gbs131.fotoapparat E/AndroidRuntime: FATAL EXCEPTION: pool-2-thread-1
                                                                                java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.lang.NullPointerException
                                                                                    at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
                                                                                    at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
                                                                                    at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
                                                                                    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080)
                                                                                    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573)
                                                                                    at java.lang.Thread.run(Thread.java:856)
                                                                                 Caused by: java.util.concurrent.ExecutionException: java.lang.NullPointerException
                                                                                    at java.util.concurrent.FutureTask.report(FutureTask.java:94)
                                                                                    at java.util.concurrent.FutureTask.get(FutureTask.java:160)
                                                                                    at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:127)
                                                                                    at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
                                                                                    at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
                                                                                    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080)
                                                                                    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573)
                                                                                    at java.lang.Thread.run(Thread.java:856)
                                                                                 Caused by: java.lang.NullPointerException
                                                                                    at io.fotoapparat.hardware.v1.Camera1.getCapabilities(Camera1.java:237)
                                                                                    at io.fotoapparat.parameter.provider.GetCapabilitiesTask$1.call(GetCapabilitiesTask.java:18)
                                                                                    at io.fotoapparat.parameter.provider.GetCapabilitiesTask$1.call(GetCapabilitiesTask.java:15)
                                                                                    at java.util.concurrent.FutureTask.run(FutureTask.java:234)
                                                                                    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080)
[... cut ...]
        fotoapparat.start();
    }

    @Override
    protected void onStop() {
        super.onStop();
        fotoapparat.stop();
    }

    public void takePicture(View v) {
        fotoapparat.takePicture()
                .toBitmap()
                .whenAvailable(new PendingResult.Callback<BitmapPhoto>() {
                    @Override
                    public void onResult(BitmapPhoto result) {
                        ...
                    }
                });
    }

**Using library version 1.2.0**

dmitry-zaitsev, 2017-08-01:
It looks like there is a racing condition and picture is being taken while the camera is stopping. Will dig deeper into that to check why it happens.

kphil, 2017-09-20:
+1 for catching these exceptions and return them to the caller via the cameraErrorCallback. Especially for calls on the camera instance like takePicture and getParameters. These often result in NPE's for no reason on some devices.

I have the same crashes with 1.3.0, for example 1) described here and the one in issue #98 .

**Exception java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.takePicture([REDACTED_DOMAIN_NAME]$ShutterCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback)' on a null object reference**
io.fotoapparat.result.PendingResult.getResultUnsafe (PendingResult.java:129)
io.fotoapparat.result.PendingResult.access$100 (PendingResult.java:20)
io.fotoapparat.result.PendingResult$2.run (PendingResult.java:98)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1115)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:590)
java.lang.Thread.run (Thread.java:818)

**Caused by java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.takePicture([REDACTED_DOMAIN_NAME]$ShutterCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback)' on a null object reference**
java.util.concurrent.FutureTask.report (FutureTask.java:93)
java.util.concurrent.FutureTask.get (FutureTask.java:163)
io.fotoapparat.result.PendingResult.getResultUnsafe (PendingResult.java:127)
io.fotoapparat.result.PendingResult.access$100 (PendingResult.java:20)
io.fotoapparat.result.PendingResult$2.run (PendingResult.java:98)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1115)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:590)
java.lang.Thread.run (Thread.java:818)

**Caused by java.util.concurrent.ExecutionException: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.takePicture([REDACTED_DOMAIN_NAME]$ShutterCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback)' on a null object reference**
java.util.concurrent.FutureTask.report (FutureTask.java:93)
java.util.concurrent.FutureTask.get (FutureTask.java:163)
io.fotoapparat.result.PendingResult$1.call (PendingResult.java:56)
java.util.concurrent.FutureTask.run (FutureTask.java:237)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1115)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:590)
java.lang.Thread.run (Thread.java:818)

**Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.takePicture([REDACTED_DOMAIN_NAME]$ShutterCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback, [REDACTED_DOMAIN_NAME]$PictureCallback)' on a null object reference**
io.fotoapparat.hardware.v1.Camera1.takePicture (Camera1.java:277)
io.fotoapparat.routine.picture.TakePictureTask$1.call (TakePictureTask.java:24)
io.fotoapparat.routine.picture.TakePictureTask$1.call (TakePictureTask.java:19)
java.util.concurrent.FutureTask.run (FutureTask.java:237)
java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1115)
java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:590)
java.lang.Thread.run (Thread.java:818)

Happened on the following devices:
### Xiaomi
- Xiaomi Redmi 2 Prime (HM 2LTE-IN)
- Redmi Note 3
- Redmi 4A
- Redmi Note 4G (HM NOTE 1LTETD)
- Mi 4i
- Redmi Note 4
- Redmi 3S
- Redmi Note 3G/4G (HM NOTE 1S CT)
- Redmi 4
- Redmi 2 (2014819)
- Redmi 2 Global (2014818)

### Oppo
- F1s (A1601)
- A33f
- A37 (A37fw-International)
- F1 Plus (X9009)
- F1 (F1w)
- A 37 (A37fw)

### Samsung
- Galaxy Note2 (GT-N7100)
- Galaxy Grand Prime (SM-G531F ,SM-G530BT)
- Galaxy Win
- Galaxy Core
- Galaxy Core Prime
- Galaxy Core Prime Duos (SM-G360H)

[... 61 lines cut ...]
                        ),
                        aspectRatio(
                                16f / 9f,
                                CameraUtils.recommendedPictureSize()
                        )))
                .previewSize(firstAvailable(
                        aspectRatio(
                                4f / 3f,
                                CameraUtils.recommendedPreviewSize()
                        ),
                        aspectRatio(
                                16f / 9f,
                                CameraUtils.recommendedPreviewSize()
                        )))
                .lensPosition(back())
                .focusMode(firstAvailable(
                        macro(),
                        continuousFocus(),
                        autoFocus(),
                        fixed()
)).flash(off())`

I solved these issues for me by catching NPEs on camera.getParameters and camera.takePicture calls, throw an CameraException and passing it later CameraErrorCallback. It's not very elegant, but if i can help i would gladly provide a PR. 

And thanks for the great camera library!

shtolik, 2017-10-25:
I see same crash reports from our users. Could the suggested fix be included in next version, please?

Diolor, 2017-10-25:
Hi @shtolik. Our time is pretty limited but we would be happy to merge a PR with the fix.

kphil, 2017-10-29:
I would provide a PR, but i would take some feedback on best way to handle errors related to routines with tasks.

Currently CameraDevice implementations are catching RuntimeExceptions and throw a CameraException from that. The CameraException is than caught in the routine (currently only in StartCameraRoutine), which also has the CameraErrorCallback. In fact CameraErrorCallback is exclusivly passed to routines.
And if the routine catches a CameraException, we call the CameraErrorCallback. So i would go on and pass the CameraErrorCallback to other routines which are using camera and are likley to cause a RuntimeException and handle it the same way.

But routines like TakePictureRoutine involve a task (FutureTask) and a result. For TakePictureRoutine i did it the same way as described before, but had to go on and handle failing to take a picture in the transformer classes (passing an empty Photo) and make PendingResults Callback onResult nullable.
Or would it be better to just handle/pass FutureTasks ExecutionException inside PendingResult if something goes wrong with a task?

Diolor, 2017-12-24:
Hey guys, you may try `2.0.0-beta2` to check if it's still happening. The crash will for sure be caught internally in FA but if there is any funky behaviour feel free to report it.

Diolor, 2017-12-29:
I am still not convinced if we should suppress these exceptions completely as the developers might want to retry an action or inform the user. 🤔

closed 2017-12-31 by Diolor

Diolor, 2018-01-03:
Just for the sake of transparency: 
- rx converters will returns those errors in the `onError`
- .await() will throw the relevant exceptions
- .whenAvailable will fail more silently and return a `null` object