RedApparat/Fotoapparat #70: Get current and available flash modes

TemMax, 2017-07-28:
How I can know which flash mode is current? And how I can all available flash modes (not from builder)?

dmitry-zaitsev, 2017-08-01:
> How I can find all available flash modes (not from builder)?

Use `Fotoapparat.getCapabilities()`, it will provide you `Capabilities` object asynchronously which will tell you which flash modes are available.

> How I can know which flash mode is current?

When you get `Capabilities` and supported flash modes you can apply the same selector function to them as you specified in builder. Like that:

```
firstAvailable(autoFlash(), torch()).select(capabilities.supportedFlashModes())
```

> And how I can switch flash mode?

Right now the only option is to create a new `Fotoapparat` instance each time you need to do that. We are thinking about improving it.

closed 2017-08-01 by dmitry-zaitsev

pflammertsma, 2017-11-22:
The ability to update camera parameters has since been included in Fotoapparat; use:

    updateParamters(UpdateRequest.builder().flash(...).build())

As for obtaining the capabilities, what wasn't immediately clear to me was that this is offered (as Dmitry said) asynchronously through a PendingResult. In Kotlin:

    fotoapparat?.capabilities?.toPendingResult()?.whenAvailable({
        it.supportedFlashModes().forEach {
            println("flash mode: " + it)
        }
    })

guyca, 2017-12-03:
I'm still not entirely sure how to get the current flash mode as `capabilities.supportedFlashModes()` lists the flash modes supported by the device and not the current mode in use.
Do I need to track this my self when changing flash modes?

Diolor, 2017-12-03:
`Fotoapparat#getCurrentParameters`