RedApparat/Fotoapparat #72: Replace appcompat-v7 Dependency with support-annotations

commonsguy, 2017-07-30:
Right now, the library depends upon `appcompat-v7`. However, the library uses nothing out of `appcompat-v7`. It does use classes from `support-annotations`, but that's it, as far as I can tell.

The workaround is to have the following in Gradle, to avoid unnecessary code and resources (in situations where the hosting app itself is not using `appcompat-v7`):

```gradle
    compile 'com.android.support:support-annotations:26.0.0'
    compile('io.fotoapparat.fotoapparat:library:1.2.0') {
      exclude group: 'com.android.support', module: 'appcompat-v7'
    }
```

Thanks for considering this!

dmitry-zaitsev, 2017-07-31:
Thanks for the input, I think it makes total sense. We'll improve that in the next release.

dmitry-zaitsev, 2017-08-07:
Done. Will be a part of 1.3.0 release.

closed 2017-08-07 by dmitry-zaitsev