RedApparat/Fotoapparat #73: oom Android 4.3,level 18 ROM  htc/D516d CPU  armeabi-v7a

MinchinWang, 2017-08-03:
1 java.lang.RuntimeException:java.util.concurrent.ExecutionException: java.lang.OutOfMemoryError
--
2 io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
3 ......
4 Caused by:
5 java.lang.OutOfMemoryError:
6 android.graphics.BitmapFactory.nativeDecodeByteArray(Native Method)
7 android.graphics.BitmapFactory.decodeByteArray(BitmapFactory.java:429)
8 android.graphics.BitmapFactory.decodeByteArray(BitmapFactory.java:447)
9 io.fotoapparat.result.transformer.BitmapPhotoTransformer.readImage(BitmapPhotoTransformer.java:53)
10 io.fotoapparat.result.transformer.BitmapPhotoTransformer.transform(BitmapPhotoTransformer.java:29)
11 io.fotoapparat.result.transformer.BitmapPhotoTransformer.transform(BitmapPhotoTransformer.java:14)
12 io.fotoapparat.result.PendingResult$1.call(PendingResult.java:55)
13 java.util.concurrent.FutureTask.run(FutureTask.java:234)
14 java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080)
15 java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573)
16 java.lang.Thread.run(Thread.java:841)

dmitry-zaitsev, 2017-08-03:
Out of memory problems are just that - not enough memory of the device. You can solve it in two ways:

- Make sure there are no memory leaks
- Add `largeHeap` flag to AndroidManifest
- Downscale the images before converting them to Bitmap (check our example)

closed 2017-08-03 by dmitry-zaitsev

zameelpichen, 2017-11-16:
"Downscale the images before converting them to Bitmap (check our example)"

can you elaborate? i couldnt find anything in the sample folder.

I am having the OOM problem.

Thanks a lot in advance.