RedApparat/Fotoapparat #77: White screen when changing orientation during launch

scifinder, 2017-08-10:
Sometimes, if you change the orientation of the screen when the application is launched, a white box appears instead of the camera. On video this is the first time: https://drive.google.com/open?id=0BzGcjhUSu0r1MFpVUWdRMGN6Z1U. Log:
E/BufferQueueProducer: [SurfaceTexture-0-17724-8] dequeueBuffer: BufferQueue has been abandoned
E/ViewRootImpl: sendUserActionEvent() returned.
(It seems that these errors always appear when changing orientation.)

dmitry-zaitsev, 2017-08-10:
That is a normal behaviour. Camera is being opened and closed asynchronously, so it is normal  that there is some delay before you see the image.

scifinder, 2017-08-11:
Yes, but sometimes the camera just does not start. Look at the first launch of the application on the video - there is a white box.

dmitry-zaitsev, 2017-08-11:
Hm, right. Sorry, forgot to check the video. I will take a look at the issue.

It would be nice if you'll provide:

- The log
- Code which you use to start and stop Fotoapparat

scifinder, 2017-08-11:
@dmitry-zaitsev Sent email.

dmitry-zaitsev, 2017-08-26:
There was indeed a racing condition which is hard to reproduce on high-end devices (if you can still call S7 Edge high end). It is fixed now and will be a part of the 1.3.0 release.

closed 2017-08-26 by dmitry-zaitsev

andrewcking, 2017-10-10:
I believe this may still be an issue, see #104 #89 

You can reproduce it by deploying the fotoapparat example to a Nexus 5 with API 22 in Android studio. Hit the home button then open the app again. You are greeted with a white camera preview that does not reinit. 

Relevant logcat:
D/Fotoapparat: setDisplaySurface
D/Camera: app passed NULL surface
D/Fotoapparat: startPreview

andrewcking, 2017-10-10:
Sorry to double comment. I think this is happening because 
'camera.setDisplaySurface(textureView);'

is being called before the runnable in updateLayout is finished. If I sleep the thread after running updateLayout but before calling that line then the issue does not occur.

```
TextureRendererView.java

    @Override
    public void attachCamera(CameraDevice camera) {
        try {
            awaitSurfaceTexture();
            updateLayout(camera);
            try{
                Thread.sleep(50);
            }catch (Exception e){
                // Do nothing
            }
            camera.setDisplaySurface(textureView);
        } catch (InterruptedException e) {
            // Do nothing
        }
    }
```

Obviously one wouldn't want to actually do that but it may help with debugging?