RedApparat/Fotoapparat #79: Error Log getSupportedPreviewSizes

REBOOTERS, 2017-08-15:
```
08-15 16:03:06.344 18901-18923/io.fotoapparat.sample E/suojp: getSupportedPictureSizes str:4160x3120,4000x3000,3840x2160,3200x2400,3120x3120,2976x2976,2592x1944,2560x1920,2688x1512,2048x1536,1920x1080,1600x1200,1280x960,1280x768,1280x720,1024x768,800x600,864x480,800x480,720x480,640x480,352x288,320x240,176x144,160x120
08-15 16:03:06.345 18901-18923/io.fotoapparat.sample E/suojp: getSupportedPreviewSizes th:java.lang.Throwable
                                                                  at android.hardware.Camera$Parameters.getSupportedPreviewSizes(Camera.java:3088)
                                                                  at io.fotoapparat.hardware.v1.capabilities.CapabilitiesFactory.extractPreviewSizes(CapabilitiesFactory.java:37)
                                                                  at io.fotoapparat.hardware.v1.capabilities.CapabilitiesFactory.fromParameters(CapabilitiesFactory.java:30)
                                                                  at io.fotoapparat.hardware.v1.Camera1.getCapabilities(Camera1.java:236)
                                                                  at io.fotoapparat.parameter.provider.InitialParametersProvider.initialParameters(InitialParametersProvider.java:45)
                                                                  at io.fotoapparat.routine.StartCameraRoutine.tryToStartCamera(StartCameraRoutine.java:58)
                                                                  at io.fotoapparat.routine.StartCameraRoutine.run(StartCameraRoutine.java:45)
                                                                  at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                                  at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                                  at java.lang.Thread.run(Thread.java:818)
```

get this error log on Android Device [letv pro1](https://www.lemall.com/le_1pro/parameter.html)

it's  resolution is  2560x1440。

**Now ,when I taka a picture ,it failed to save it in Android/data/io/io.fotoapparat.sample/files/photos/  ,I'm not sure wheather this error log lead to fail save picture .**.How could i solve this problem.forgive my poor english.

dmitry-zaitsev, 2017-08-26:
Could you please show me how you initialize `Fotoapparat` instance and what are you using for picture size and for preview size.

REBOOTERS, 2017-08-27:
Well ,forget to say.I clone the code from master branch and run the application get this problem.
On other phone xiaomi5  run the same application without any problem.

Diolor, 2017-12-13:
I expect here that the device implementation is broken since `getSupportedPreviewSizes` is not expected to return any exception. Only think we could do here to support it is to leave `PreviewSizes` untouched which I don't really like

Diolor, 2017-12-24:
So, this device is violating the android API contract meaning that the device is not returning any preview sizes. I propose we don't support this device. Pretty sure it's users have other apps which are broken too.

Info, Android jdoc for `getSupportedPreviewSizes`:

```java
        /**
         * Gets the supported preview sizes.
         *
         * @return a list of Size object. This method will always return a list
         *         with at least one element.
         */
```

closed 2017-12-24 by Diolor