RedApparat/Fotoapparat #80: java.util.concurrent.ExecutionException

sergey-shpital, 2017-08-15:
# Platform: android
# OS Version: 6.0.1
# Device: MI 5
# RAM Free: 16.7%
# Disk Free: 43.7%

#0. Crashed: pool-9-thread-1: 0 0 0x0000000000000000
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

--

Fatal Exception: java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
       at java.util.concurrent.FutureTask.report(FutureTask.java:94)
       at java.util.concurrent.FutureTask.get(FutureTask.java:164)
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:127)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.util.concurrent.ExecutionException: java.lang.RuntimeException: takePicture failed
       at java.util.concurrent.FutureTask.report(FutureTask.java:94)
       at java.util.concurrent.FutureTask.get(FutureTask.java:164)
       at io.fotoapparat.result.PendingResult$1.call(PendingResult.java:56)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)
Caused by java.lang.RuntimeException: takePicture failed
       at android.hardware.Camera.native_takePicture(Camera.java)
       at android.hardware.Camera.takePicture(Camera.java:1710)
       at io.fotoapparat.hardware.v1.Camera1.takePicture(Camera1.java:258)
       at io.fotoapparat.routine.picture.TakePictureTask$1.call(TakePictureTask.java:24)
       at io.fotoapparat.routine.picture.TakePictureTask$1.call(TakePictureTask.java:19)
       at java.util.concurrent.FutureTask.run(FutureTask.java:237)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

#0. Crashed: pool-9-thread-1: 0 0 0x0000000000000000
       at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
       at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
       at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at java.lang.Thread.run(Thread.java:818)

#1. AsyncTask #4
       at android.graphics.Bitmap.nativeCompress(Bitmap.java)
       at android.graphics.Bitmap.compress(Bitmap.java:1039)
       at club.rentmee.v2.AFileSendHttpRequester.generateHttpPostParams(AFileSendHttpRequester.java:60)
       at club.rentmee.v2.AFileSendHttpRequester.generateRequest(AFileSendHttpRequester.java:37)
       at club.rentmee.v2.AFileSendHttpRequester.generateRequest(AFileSendHttpRequester.java:18)
       at club.rentmee.v2.HTTPRequester.doAsyncRequest(HTTPRequester.java:170)
       at club.rentmee.v2.HTTPRequester.access$000(HTTPRequester.java:32)
       at club.rentmee.v2.HTTPRequester$1.doInBackground(HTTPRequester.java:120)
       at club.rentmee.v2.HTTPRequester$1.doInBackground(HTTPRequester.java:115)
       at android.os.AsyncTask$2.call(AsyncTask.java:295)
[... 480 lines cut ...]
       at java.lang.Thread.run(Thread.java:818)

#49. ReferenceQueueDaemon
       at java.lang.Object.wait(Object.java)
       at java.lang.Daemons$ReferenceQueueDaemon.run(Daemons.java:147)
       at java.lang.Thread.run(Thread.java:818)

#50. GAC_Executor[0]
       at java.lang.Object.wait(Object.java)
       at java.lang.Thread.parkFor$(Thread.java:1220)
       at sun.misc.Unsafe.park(Unsafe.java:299)
       at java.util.concurrent.locks.LockSupport.park(LockSupport.java:158)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2013)
       at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:410)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1036)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1098)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
       at com.google.android.gms.internal.zzrn.run(Unknown Source)
       at java.lang.Thread.run(Thread.java:818)

#51. Okio Watchdog
       at java.lang.Object.wait(Object.java)
       at com.android.okhttp.okio.AsyncTimeout.awaitTimeout(AsyncTimeout.java:311)
       at com.android.okhttp.okio.AsyncTimeout.access$000(AsyncTimeout.java:40)
       at com.android.okhttp.okio.AsyncTimeout$Watchdog.run(AsyncTimeout.java:286)

sergey-shpital, 2017-08-15:
This error also exists on  OnePlus OnePlus3T (OnePlus3T), 6144MB RAM, Android 7.1

dmitry-zaitsev, 2017-08-20:
It is hard to tell what is going wrong exactly since Camera API gives us little to no information about the crash, but I would assume that Camera was not able to allocate enough memory for your image. Possibly because you are using another `Bitmap` at the same time (I spotted an `AsyncTask` in one of your threads).

Things to try:

- Use `largeHeap` flag in your `AndroidManifest`. That is advisable for apps which are working with images.
- Try to postpone heavy images operations until a user is done with the camera. Or vice versa - do not allow the user to take new photos while there is an ongoing image processing. We are working with limited mobile resources after all.

closed 2017-08-20 by dmitry-zaitsev

sergey-shpital, 2017-10-02:
despite all the effort, quite a frequent mistake :(

OnePlus OnePlus3T (OnePlus3T), 6144MB RAM, Android 7.1

`java.lang.RuntimeException: 
  at io.fotoapparat.result.PendingResult.getResultUnsafe (PendingResult.java:129)
  at io.fotoapparat.result.PendingResult.access$100 (PendingResult.java:20)
  at io.fotoapparat.result.PendingResult$2.run (PendingResult.java:98)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1133)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:607)
  at java.lang.Thread.run (Thread.java:761)
Caused by: java.util.concurrent.ExecutionException: 
  at java.util.concurrent.FutureTask.report (FutureTask.java:94)
  at java.util.concurrent.FutureTask.get (FutureTask.java:164)
  at io.fotoapparat.result.PendingResult.getResultUnsafe (PendingResult.java:127)`