RedApparat/Fotoapparat #86: NullPointerException Camera1.getCapabilities

n1n3b1t, 2017-08-30:
It happens on Android 4,5,6, mostly Chinese devices - xiaomi, oppo, Huawei
Fatal Exception: java.lang.NullPointerException
       at io.fotoapparat.hardware.v1.Camera1.getCapabilities(SourceFile:237)
       at io.fotoapparat.routine.parameter.UpdateParametersRoutine.updateParameters(SourceFile:32)
       at io.fotoapparat.Fotoapparat$1.run(SourceFile:228)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
       at java.lang.Thread.run(Thread.java:841)

jpribble, 2017-12-07:
I observed a similar exception on a HTC One X+ running Android 4.2.2 right after opening the camera. This seems to occur very rarely (about 1 in 100 launches). Please let me know if you think it's related to the issue above or if this should be tracked separately. Here's the stack trace:

12-07 17:07:01.487 8035-8695/capitalone.com.goviddemo E/AndroidRuntime: FATAL EXCEPTION: pool-3-thread-1
java.lang.RuntimeException: java.util.concurrent.ExecutionException: java.lang.NullPointerException
    at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:129)
    at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20)
    at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573)
    at java.lang.Thread.run(Thread.java:864)
 Caused by: java.util.concurrent.ExecutionException: java.lang.NullPointerException
    at java.util.concurrent.FutureTask.report(FutureTask.java:94)
    at java.util.concurrent.FutureTask.get(FutureTask.java:160)
    at io.fotoapparat.result.PendingResult.getResultUnsafe(PendingResult.java:127)
    at io.fotoapparat.result.PendingResult.access$100(PendingResult.java:20) 
    at io.fotoapparat.result.PendingResult$2.run(PendingResult.java:98) 
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080) 
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573) 
    at java.lang.Thread.run(Thread.java:864) 
 Caused by: java.lang.NullPointerException
    at io.fotoapparat.hardware.v1.Camera1.getCapabilities(Camera1.java:260)
    at io.fotoapparat.parameter.provider.GetCapabilitiesTask$1.call(GetCapabilitiesTask.java:18)
    at io.fotoapparat.parameter.provider.GetCapabilitiesTask$1.call(GetCapabilitiesTask.java:15)
    at java.util.concurrent.FutureTask.run(FutureTask.java:234)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1080) 
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:573) 
    at java.lang.Thread.run(Thread.java:864) 

Can we wrap all calls to cameraDevice.getCapabilities() in a try/catch since the caller can't catch it inside the FutureTask?

Diolor, 2017-12-13:
Hmm, might be different. I honestly don't know. Maybe `getCapabilities` runs before `start` somewhow. I have created a `v2` branch which creates the capabilities always after start and will crash more gratefully

Diolor, 2017-12-31:
If capabilities cannot be obtained we will return an empty value. If you have a better idea, feel free to comment

closed 2017-12-31 by Diolor