RedApparat/Fotoapparat #88: Shoot the photos and start the preview again very slowly

xiaolongyuan, 2017-09-07:


xiaolongyuan, 2017-09-07:
```
    String savePath = Environment.getExternalStoragePublicDirectory
                (Environment
                        .DIRECTORY_DCIM) + "/CAMERA/";

        File dir = new File(savePath);
        if (!dir.exists()) {
            dir.mkdir();
        }

        final String path = savePath + "IMG_" + new Date().getTime() + ".jpg";
        final File file = new File(path);
        final PhotoResult photoResult = fotoapparat.takePicture();
        return Observable
                .create(new Observable.OnSubscribe<String>() {
                    @Override
                    public void call(final Subscriber<? super String> subscriber) {

                        Logger.i("此时线程："+Thread.currentThread().getName());
                        photoResult.saveToFile(file)
                                .whenAvailable(new PendingResult.Callback<Void>() {
                                    @Override
                                    public void onResult(Void aVoid) {
                                        Logger.i("whenAvailable线程："+Thread.currentThread().getName());
                                        subscriber.onNext(path);
                                        subscriber.onCompleted();
                                    }
                                });
                        ;
                    }
                })
                .subscribeOn(Schedulers.io())
                ;
```

It takes 8-10 seconds to start previewing

faketon, 2017-09-08:
i think image is big. i use big image some device is slow and some device is normal. are you try photoSize(smallestSize).

dmitry-zaitsev, 2017-10-03:
There is nothing we can do about that - that fully depends on capabilities of the device.

By the way, for using Fotoapparat with RxJava consider using our adapters: https://github.com/Fotoapparat/Fotoapparat/tree/master/fotoapparat-adapters

closed 2017-10-03 by dmitry-zaitsev