RedApparat/Fotoapparat #89: Preview screen is not displayed(black) when start is called again

nicekei, 2017-09-09:
Hi,
I'm testing this device on Galaxy note 4. (os version 6.0.1)
All test is done in landscape mode.

09-10 07:18:04.718 22918-23016/com.m2s.eyetoolone D/Fotoapparat: close
09-10 07:18:04.908 22918-22918/com.m2s.eyetoolone V/ActivityThread: updateVisibility : ActivityRecord{636317 token=android.os.BinderProxy@9fa6f2d {com.m2s.eyetoolone/com.m2s.eyetoolone.MainActivity}} show : false
09-10 07:18:09.468 22918-23016/com.m2s.eyetoolone D/Fotoapparat: open
09-10 07:18:09.473 22918-22918/com.m2s.eyetoolone D/SensorManager: registerListener :: 0, ICM20610 Acceleration Sensor, 200000, 0,  
09-10 07:18:09.483 22918-23016/com.m2s.eyetoolone D/Fotoapparat: getCapabilities
09-10 07:18:09.488 22918-23016/com.m2s.eyetoolone D/Fotoapparat: updateParameters
09-10 07:18:09.493 22918-23016/com.m2s.eyetoolone D/Fotoapparat: setDisplayOrientation
09-10 07:18:09.493 22918-23016/com.m2s.eyetoolone D/Fotoapparat: getRendererParameters
09-10 07:18:09.493 22918-23016/com.m2s.eyetoolone D/Fotoapparat: Renderer parameters are: RendererParameters{previewSize=Size{width=1920, height=1080}, frameRotation=0}
09-10 07:18:09.493 22918-23016/com.m2s.eyetoolone D/Fotoapparat: setDisplaySurface
09-10 07:18:09.493 22918-23016/com.m2s.eyetoolone D/Camera: app passed NULL surface
09-10 07:18:09.493 22918-23016/com.m2s.eyetoolone D/Fotoapparat: startPreview
09-10 07:18:09.513 22918-23021/com.m2s.eyetoolone D/mali_winsys: new_window_surface returns 0x3000,  [2560x1440]-format:1
09-10 07:18:09.563 22918-22918/com.m2s.eyetoolone I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@9fa6f2d time:482124066

- When the app is turned to foreground and start() is called again, the preview screen is not displayed. But taking photo works in this no preview state.
- If I switch to other app, and return to the test app, the preview is displaying again.
- If I turn off the phone, and turn on. It was once recovered, but the screen was stretched to the center.

ekzee, 2017-09-18:
We experiencing similar issue. Calling:
```
camera.start();
camera.stop();
camera.start();
```
in onResume() seems to fix the issue, but adds quite a nasty lag to activity start.

kerimovscreations, 2017-09-21:
If, some relevant solution will be found, please mention me here.

andrewcking, 2017-10-10:
Seeing this as well on Nexus 5 5.1.1 (api 22)

Diolor, 2017-12-31:
Hey there, please test v2 and if you still experience it, let us know

closed 2017-12-31 by Diolor

kerimovscreations, 2018-01-03:
@Diolor thanks for the response, unfortunately, the v2 library doesn't work properly in Java. **firstAvailable** method is not recognizable

ErAmrit, 2019-08-05:
No need to call given below code inside onResume() it works inside onStart() also, but the number of phones works well without this nasty code. I think there are many phones which runs on customized Android OS on that normal flow not works properly. 

{
camera.start();
camera.stop();
camera.start();
}
@ekzee @Diolor