RedApparat/Fotoapparat #95: Camera1 preview is darker comparing to stock camera app

friendoye, 2017-09-18:
I don't know how is it possible, but Fotoapparat preview seems darker comparing to stock camera app or another camera lib ([Camera-Kit](https://github.com/gogopop/CameraKit-Android)).

Seems like it's not depend on Android version or manufacture, but if you would like to get more info I'll add more detailed explanation of the problem.

friendoye, 2017-09-18:
After my small investigation I've figured out, that using higher FPS for ```previewFpsRange()``` causes frames be more darker.

Diolor, 2017-10-03:
Interesting to know. Should we close this issue as it's "expected behavior" from the camera API or the hardware?

dmitry-zaitsev, 2017-10-03:
I believe it is an expected behavior. Also you might notice that other camera apps are cranking up the brightness of the screen when you open them.

closed 2017-10-03 by dmitry-zaitsev

valterh4ck3r, 2018-02-13:
How to change brightness camera ?

Diolor, 2018-02-13:
Try remove `.sensorSensitivity(highestSensorSensitivity())`

friendoye, 2018-02-13:
From my experience, setting ```previewFpsRange(lowestFps())``` should make preview brighter.

valterh4ck3r, 2018-02-13:
Really, it's true... Thanks @friendoye