RedApparat/Fotoapparat #96: Сontinuous Focus dont restored after takePicture() on Xiaomi Redmi 4x

DmitryMtv, 2017-09-18:
I set the focus mode:        
```
return Fotoapparat
                .with(this)
.....
                .focusMode( firstAvailable(
                        continuousFocus()
                ))
.....
```
After calling `fotoapparatSwitcher.getCurrentFotoapparat().takePicture();` focus mode dont restored to continuous mode and not working. I found a solution, switch twice focus mode, but it's not very nice.
  ```
  private void takePicture() {
        PhotoResult photoResult = fotoapparatSwitcher.getCurrentFotoapparat().takePicture();

        photoResult.saveToFile(new File(
                getExternalFilesDir("photos"),
                "photo.jpg"
        ));
...
       fotoapparatSwitcher
                .getCurrentFotoapparat()
                .updateParameters(
                        UpdateRequest.builder()
                                .focusMode(firstAvailable(fixed()
                                ))
                                .build()
                );
        fotoapparatSwitcher
                .getCurrentFotoapparat()
                .updateParameters(
                        UpdateRequest.builder()
                                .focusMode(firstAvailable(continuousFocus()
                                ))
                                .build()
                );
    }
```
Perhaps you will tell us the more correct solution?  I think this is a problem of a specific model.

dmitry-zaitsev, 2017-10-03:
I can reproduce your issue. Somehow I didn't notice that this happens... will try to fix it asap.

dmitry-zaitsev, 2017-10-03:
The issue should be fixed now. Please use new version of the library: 1.4.1

closed 2017-10-03 by dmitry-zaitsev

DmitryMtv, 2017-10-12:
A similar situation occurs if the camera does not have time to focus. Ie, if in Continuous mode I call takePicture () at the moment when the camera did not have time to focus, then the Continuous mode stops working.