RedApparat/Fotoapparat #100: Photo Capture is taking too long time

shivam2702, 2017-09-22:
```
photoResult = fotoapparat.takePicture();
        photoResult
                .toBitmap()
                .whenAvailable(new PendingResult.Callback<BitmapPhoto>() {
                    @Override
                    public void onResult(BitmapPhoto result) {
                        selectedBitmap = rotateBitmap(result.bitmap, -result.rotationDegrees);
                        stopCamera();
                    }
                });
```
```
I/Choreographer:  Skipped 94 frames!  The application may be doing too much work on its main thread.
D/Fotoapparat: stopPreview
I/Choreographer: Skipped 39 frames!  The application may be doing too much work on its main thread.

```
I am using version 1.3.0
Please let me know if I am doing anything wrong

dmitry-zaitsev, 2017-09-26:
The reason why it takes a while is that you are rotating the image on the main thread. That is not a particularly fast operation, so consider moving it to a background thread.

One other (better) option would be to use `Transformer`:

```java
photoResult
                .toBitmap()
                .transform(new Transformer<BitmapPhoto, Bitmap>() {
                    @Override
                    public Bitmap transform(BitmapPhoto input) {
                        return rotateBitmap(input.bitmap, -input.rotationDegrees);
                    }
                })
                .whenAvailable(new PendingResult.Callback<Bitmap>() {
                    @Override
                    public void onResult(Bitmap result) {
                        selectedBitmap = result;
                        stopCamera();
                    }
                });
```

closed 2017-09-26 by dmitry-zaitsev

shivam2702, 2017-09-27:
Thank's it saves time.
but still it's taking approx 2 sec of time.
Is it normal?

dmitry-zaitsev, 2017-09-27:
Yes, that is expected. Most of the time goes into decoding `Bitmap` from JPEG byte array which camera sends to us. You can speed up the process by reducing the picture size in Fotoapparat initialization.