RedApparat/Fotoapparat #101: Errors when onBackPressed

Paul75, 2017-09-24:
Hello,

The message in console is :
`09-24 18:20:48.055 8021-8036/fr.laposte.gompost E/BufferQueueProducer: [SurfaceTexture-0-8021-0] dequeueBuffer: BufferQueue has been abandoned
09-24 18:20:48.063 8021-8034/fr.laposte.gompost E/BufferQueueProducer: [SurfaceTexture-0-8021-0] queueBuffer: BufferQueue has been abandoned
09-24 18:20:48.096 8021-8036/fr.laposte.gompost E/BufferQueueProducer: [SurfaceTexture-0-8021-0] queueBuffer: BufferQueue has been abandoned
09-24 18:20:48.130 8021-8034/fr.laposte.gompost E/BufferQueueProducer: [SurfaceTexture-0-8021-0] queueBuffer: BufferQueue has been abandoned
09-24 18:20:48.163 8021-8446/fr.laposte.gompost E/BufferQueueProducer: [SurfaceTexture-0-8021-0] queueBuffer: BufferQueue has been abandoned`

when I do a onBackPressed.

Normal ?

Diolor, 2017-09-24:
Are you using camera v1 or v2?

Paul75, 2017-09-24:
I'm using the code that have in the source code.

Diolor, 2017-09-24:
Okay, in other words. How do initialize FA and when do you call .onStop()?

Paul75, 2017-09-24:
Hello,

I call ::
`private FotoapparatSwitcher fotoapparatSwitcher;
    private Fotoapparat frontFotoapparat;

cameraView = findViewById(R.id.camera_view);

private void setupFotoapparat() {
        backFotoapparat = createFotoapparat(LensPosition.BACK);
        fotoapparatSwitcher = FotoapparatSwitcher.withDefault(backFotoapparat);
    }

@Override
    protected void onStart() {
        super.onStart();
        if (hasCameraPermission) {
            fotoapparatSwitcher.start();
        }
    }

    @Override
    protected void onStop() {
        super.onStop();
        if (hasCameraPermission) {
            fotoapparatSwitcher.stop();
        }
    }

@Override
    public void onBackPressed() {
        super.onBackPressed();
        finish();
    }

`

Diolor, 2017-09-24:
Thanks, but could you please post the full relevant code? E.g. What is in the `createFotoapparat` method? Also, is this happening on some specific device/API version? Thanks!

Paul75, 2017-09-24:
Hello,

API 24.

Code ::

`package fr.laposte.gompost.activity.photos;

import android.Manifest;
import android.content.pm.PackageManager;
import android.os.Environment;
import android.support.annotation.NonNull;
import android.support.v4.content.ContextCompat;
import android.support.v7.app.AppCompatActivity;
import android.os.Bundle;
import android.support.v7.widget.SwitchCompat;
import android.view.View;
import android.widget.CompoundButton;
import android.widget.ImageView;
import android.widget.SeekBar;
import android.widget.Toast;

import java.io.File;

import fr.laposte.gompost.R;
import io.fotoapparat.Fotoapparat;
import io.fotoapparat.FotoapparatSwitcher;
import io.fotoapparat.error.CameraErrorCallback;
import io.fotoapparat.hardware.CameraException;
import io.fotoapparat.parameter.LensPosition;
import io.fotoapparat.parameter.ScaleType;
import io.fotoapparat.parameter.update.UpdateRequest;
import io.fotoapparat.photo.BitmapPhoto;
import io.fotoapparat.preview.Frame;
import io.fotoapparat.preview.FrameProcessor;
import io.fotoapparat.result.PendingResult;
import io.fotoapparat.result.PhotoResult;
import io.fotoapparat.view.CameraView;

import static io.fotoapparat.log.Loggers.fileLogger;
import static io.fotoapparat.log.Loggers.logcat;
import static io.fotoapparat.log.Loggers.loggers;
import static io.fotoapparat.parameter.selector.AspectRatioSelectors.standardRatio;
import static io.fotoapparat.parameter.selector.FlashSelectors.autoFlash;
import static io.fotoapparat.parameter.selector.FlashSelectors.autoRedEye;
import static io.fotoapparat.parameter.selector.FlashSelectors.off;
import static io.fotoapparat.parameter.selector.FlashSelectors.torch;
import static io.fotoapparat.parameter.selector.FocusModeSelectors.autoFocus;
import static io.fotoapparat.parameter.selector.FocusModeSelectors.continuousFocus;
import static io.fotoapparat.parameter.selector.FocusModeSelectors.fixed;
import static io.fotoapparat.parameter.selector.LensPositionSelectors.lensPosition;
import static io.fotoapparat.parameter.selector.Selectors.firstAvailable;
import static io.fotoapparat.parameter.selector.SizeSelectors.biggestSize;
import static io.fotoapparat.result.transformer.SizeTransformers.scaled;

public class CameraActivity extends AppCompatActivity {
    private final PermissionsDelegate permissionsDelegate = new PermissionsDelegate(this);
    private boolean hasCameraPermission;
    private CameraView cameraView;

    private FotoapparatSwitcher fotoapparatSwitcher;
    private Fotoapparat frontFotoapparat;
    private Fotoapparat backFotoapparat;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_camera);

        cameraView = findViewById(R.id.camera_view);
[... 177 lines cut ...]
                                           @NonNull String[] permissions,
                                           @NonNull int[] grantResults) {
        super.onRequestPermissionsResult(requestCode, permissions, grantResults);
        if (permissionsDelegate.resultGranted(requestCode, permissions, grantResults)) {
            fotoapparatSwitcher.start();
            cameraView.setVisibility(View.VISIBLE);
        }
    }

    private class SampleFrameProcessor implements FrameProcessor {

        @Override
        public void processFrame(Frame frame) {
            // Perform frame processing, if needed
        }

    }

    @Override
    public void onBackPressed() {
        super.onBackPressed();
        finish();
    }
}
`

dmitry-zaitsev, 2017-09-26:
I can't say what is the exact reason for that, but I can say that this is a normal behaviour which is present in other camera libraries as well. It does not affect the app in any harmful way.

closed 2017-09-26 by dmitry-zaitsev