RedApparat/Fotoapparat #102: OnTouchListener Not Working

eygraber, 2017-09-26:
It's not possible to set a touch listener on `CameraView` because the focus layout steals touches. 

My use case is to take a picture when the `CameraView` is tapped. I've gotten around it by putting a transparent overlay on top, and setting the touch listener on that, but I'd rather not have to.

dmitry-zaitsev, 2017-09-26:
I can't reproduce this issue in our sample app. Also, not sure which focus layout are you talking about?

I am closing this issue until further information is provided.

closed 2017-09-26 by dmitry-zaitsev

danielgindi, 2019-07-11:
In `FocusView.kt`:
```
@SuppressLint("ClickableViewAccessibility")
override fun onTouchEvent(event: MotionEvent): Boolean {
  tapDetector.onTouchEvent(event)
  return true // <------ This line, means "the event was consumed". Thus the parent view does not get any touches
}
```