RedApparat/Fotoapparat #103: Don't output the data array in toString in Frame.

Floens, 2017-09-26:
The `toString` method of `io.fotoapparat.preview.Frame` returns the content of the image byte array. This results in the return value of the toString method to be enormous. This is slowing down my debugger to the point that it's unusable, because it has to download megabytes of string data. That, and I don't think there's any reason to actually output this data in toString.

#### What are you trying to achieve or the steps to reproduce?

I'm debugging some code that has variable of type Frame. As a result, the debugger calls toString on the frame to show it in the debugger screen, and the debugger is slowed down.

#### Context:

* *FA version*: 1.3.0
* *Devices/APIs affected*: N/A
* *any other relevant information*: N/A

dmitry-zaitsev, 2017-09-26:
We don't have a particular reason for returning the data the way we do, so we can fix the issue for you.

dmitry-zaitsev, 2017-09-26:
That is done. Will be a part of the next release (which you can expect until the end of this week).

If you need the fix right now, you can use `develop-SNAPSHOT` as a version name - that would give you the latest state we have in `develop`.

closed 2017-09-26 by dmitry-zaitsev