RedApparat/Fotoapparat #104: Screen white after rotating during picture capture

mickod, 2017-09-26:
Using the sample app with no modifications, cloned yesterday from main branch, on a Nokia 5 phone:

If you tap the screen to take a picture and then rotate the device the screen goes blank white with no preview, but with the zoom and flash buttons still displayed.

Rotating back does not bring back the preview.

It does not happen every time but does not take more than 2 or 3 attempts before it happens.

Logcat during just this sequence:

09-26 19:31:31.638 10447-10447/io.fotoapparat.sample V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@24489dd
09-26 19:31:31.800 10447-10465/io.fotoapparat.sample D/Fotoapparat: autoFocus
09-26 19:31:32.095 10447-10465/io.fotoapparat.sample D/Fotoapparat: takePicture
09-26 19:31:33.883 10447-10465/io.fotoapparat.sample D/Fotoapparat: startPreview
09-26 19:31:33.973 10447-10633/io.fotoapparat.sample I/ExifInterface_JNI: Raw image not detected
09-26 19:31:33.976 10447-10633/io.fotoapparat.sample W/ExifInterface: Skip the tag entry since tag number is not defined: 2
09-26 19:31:35.948 10447-10465/io.fotoapparat.sample D/Fotoapparat: autoFocus
09-26 19:31:36.234 10447-10465/io.fotoapparat.sample D/Fotoapparat: takePicture
09-26 19:31:36.799 10447-10447/io.fotoapparat.sample V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@953649b
09-26 19:31:36.800 10447-10447/io.fotoapparat.sample V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@c0b4038
09-26 19:31:37.336 10447-10459/io.fotoapparat.sample E/BufferQueueProducer: [SurfaceTexture-0-10447-0] queueBuffer: BufferQueue has been abandoned
09-26 19:31:37.547 10447-10465/io.fotoapparat.sample D/Fotoapparat: startPreview
09-26 19:31:37.549 10447-10465/io.fotoapparat.sample E/BufferQueueProducer: [SurfaceTexture-0-10447-0] query: BufferQueue has been abandoned
09-26 19:31:37.550 10447-10465/io.fotoapparat.sample D/Fotoapparat: setDisplayOrientation
09-26 19:31:37.556 10447-10465/io.fotoapparat.sample D/Fotoapparat: stopPreview
09-26 19:31:37.562 10447-10633/io.fotoapparat.sample I/ExifInterface_JNI: Raw image not detected
09-26 19:31:37.562 10447-10633/io.fotoapparat.sample W/ExifInterface: Skip the tag entry since tag number is not defined: 2
09-26 19:31:37.753 10447-10465/io.fotoapparat.sample D/Fotoapparat: close
09-26 19:31:38.141 10447-10465/io.fotoapparat.sample D/Fotoapparat: open
09-26 19:31:38.293 10447-10465/io.fotoapparat.sample D/Fotoapparat: getCapabilities
                                                                    
                                                                    [ 09-26 19:31:38.295   920:10937 E/         ]
                                                                    imx258_at24c64d_main_eeprom_get_calibration_items:71is_wbc:1,is_afc:1,is_lsc:1,is_dpc:0,is_insensor:0,is_ois:0
09-26 19:31:38.299 10447-10465/io.fotoapparat.sample D/Fotoapparat: updateParameters
                                                                    
                                                                    [ 09-26 19:31:38.302   920:10937 E/         ]
                                                                    imx258_at24c64d_main_autofocus_calibration:615Enter
09-26 19:31:38.303 10447-10465/io.fotoapparat.sample D/Fotoapparat: setDisplayOrientation
09-26 19:31:38.307 10447-10465/io.fotoapparat.sample D/Fotoapparat: getRendererParameters
09-26 19:31:38.308 10447-10465/io.fotoapparat.sample D/Fotoapparat: Renderer parameters are: RendererParameters{previewSize=Size{width=1600, height=1200}, frameRotation=0}
09-26 19:31:38.309 10447-10465/io.fotoapparat.sample D/Fotoapparat: setDisplaySurface
09-26 19:31:38.310 10447-10465/io.fotoapparat.sample D/Fotoapparat: startPreview
09-26 19:31:38.317 10447-10465/io.fotoapparat.sample D/Fotoapparat: getPreviewStream
09-26 19:31:38.657 10447-10465/io.fotoapparat.sample D/Fotoapparat: setDisplayOrientation
09-26 19:31:40.213 10447-10465/io.fotoapparat.sample D/Fotoapparat: autoFocus
09-26 19:31:40.528 10447-10465/io.fotoapparat.sample D/Fotoapparat: takePicture
09-26 19:31:41.070 10447-10460/io.fotoapparat.sample E/BufferQueueProducer: [SurfaceTexture-0-10447-1] cancelBuffer: BufferQueue has been abandoned
09-26 19:31:41.071 10447-10531/io.fotoapparat.sample E/BufferQueueProducer: [SurfaceTexture-0-10447-1] cancelBuffer: BufferQueue has been abandoned
09-26 19:31:41.074 10447-10460/io.fotoapparat.sample E/BufferQueueProducer: [SurfaceTexture-0-10447-1] cancelBuffer: BufferQueue has been abandoned
09-26 19:31:41.091 10447-10447/io.fotoapparat.sample V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@7e74c7b
09-26 19:31:41.091 10447-10447/io.fotoapparat.sample V/BoostFramework: BoostFramework() : mPerf = com.qualcomm.qti.Performance@8eb3a98
09-26 19:31:41.236 10447-10460/io.fotoapparat.sample E/BufferQueueProducer: [SurfaceTexture-0-10447-1] query: BufferQueue has been abandoned

mickod, 2017-10-16:
Note just in case it is not clear in the original description above - to reproduce tap to take a photo and then immediately rotate the deice, i.e. while the photo is being taken.

Diolor, 2017-12-24:
When we rotate the activity, a new instance of FA is created but we use the same thread to both shutdown the previous FA and take the picture which results in a deadlock. 

In order to solve this from now on, if stop is called all the photos which have not yet been delivered will be lost.

Diolor, 2017-12-31:
Fixed in v2.

closed 2017-12-31 by Diolor