RedApparat/Fotoapparat #105: RuntimeException set display orientation failed

chucky86boy, 2017-09-29:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

This doesn't happen overtime. It happened for this particular case while I was testing.

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
mFotoapparat = Fotoapparat
                .with(getActivity())
                .into(mCameraView)
                .photoSize(smallestSize())
                .lensPosition(front())
                .focusMode(firstAvailable(
                        autoFocus(),
                        fixed()
                ))
                .frameProcessor(mViewModel.getImageStreamPublishSubject()::onNext)
                .build();

@Override
    public void onStart() {
        super.onStart();
        mStarted = true;
        mFotoapparat.start();
        ....
    }
```

#### What was the result you received?
Crash

#### What did you expect?
When camera opens up, the bug happened

#### Context:
compile 'io.fotoapparat.fotoapparat:library:1.3.0'

# Crashlytics - plaintext stacktrace downloaded by Chuck Won at Fri, 29 Sep 2017 02:23:33 GMT
# URL: https://fabric.io/taghive/android/apps/com.tag_hive.tagplus.debug/issues/59cda700be077a4dcc6f015d?time=last-twenty-four-hours/sessions/59CDA6A301AA0001765D77ABDC9366F2_a25de715a4b811e78bd256847afe9799_0_v2
# Organization: Tag-Hive
# Platform: android
# Application: Tag Plus
# Version: 0.0.4 (1)
# Bundle Identifier: com.tag_hive.tagplus.debug
# Issue #: 108
# Issue ID: 59cda700be077a4dcc6f015d
# Session ID: 59CDA6A301AA0001765D77ABDC9366F2_a25de715a4b811e78bd256847afe9799_0_v2
# Date: 2017-09-29T01:50:00Z
# OS Version: 7.0
# Device: Galaxy Tab S2 8.0
# RAM Free: 37.2%
# Disk Free: 78.8%

#0. Crashed: pool-10-thread-1: 0 0 0x0000000000000000
       at android.hardware.Camera.setDisplayOrientation(Camera.java)
       at io.fotoapparat.hardware.v1.Camera1.setDisplayOrientation(Camera1.java:190)
       at io.fotoapparat.routine.UpdateOrientationRoutine$1.run(UpdateOrientationRoutine.java:44)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
       at java.lang.Thread.run(Thread.java:762)

--
[... 642 lines cut ...]
       at sun.misc.Unsafe.park(Unsafe.java:325)
       at java.util.concurrent.locks.LockSupport.park(LockSupport.java:161)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2035)
       at java.util.concurrent.PriorityBlockingQueue.take(PriorityBlockingQueue.java:519)
       at io.fabric.sdk.android.services.concurrency.DependencyPriorityBlockingQueue.performOperation(DependencyPriorityBlockingQueue.java:197)
       at io.fabric.sdk.android.services.concurrency.DependencyPriorityBlockingQueue.get(DependencyPriorityBlockingQueue.java:236)
       at io.fabric.sdk.android.services.concurrency.DependencyPriorityBlockingQueue.take(DependencyPriorityBlockingQueue.java:65)
       at io.fabric.sdk.android.services.concurrency.DependencyPriorityBlockingQueue.take(DependencyPriorityBlockingQueue.java:46)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1058)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1118)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
       at java.lang.Thread.run(Thread.java:762)

#55. GAC_Executor[1]
       at java.lang.Object.wait(Object.java)
       at java.lang.Thread.parkFor$(Thread.java:2128)
       at sun.misc.Unsafe.park(Unsafe.java:325)
       at java.util.concurrent.locks.LockSupport.park(LockSupport.java:161)
       at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2035)
       at java.util.concurrent.LinkedBlockingQueue.take(LinkedBlockingQueue.java:413)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1058)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1118)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
       at com.google.android.gms.internal.zzbic.run(Unknown Source)
       at java.lang.Thread.run(Thread.java:762)

jpribble, 2017-10-24:
I observed the same crash while running the sample app (version 1.4.1) on a Samsung GS6 (Android 6.0.1). Here's the stack trace:

io.fotoapparat.sample E/AndroidRuntime: FATAL EXCEPTION: pool-2-thread-1
                                                                       Process: io.fotoapparat.sample, PID: 31819
                                                                       java.lang.RuntimeException: set display orientation failed
                                                                           at android.hardware.Camera.setDisplayOrientation(Native Method)
                                                                           at io.fotoapparat.hardware.v1.Camera1.setDisplayOrientation(Camera1.java:190)
                                                                           at io.fotoapparat.routine.UpdateOrientationRoutine$1.run(UpdateOrientationRoutine.java:44)
                                                                           at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                           at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                           at java.lang.Thread.run(Thread.java:818)

Diolor, 2017-12-02:
Fixed by #126 and 1.5.0. Thanks @jpribble

closed 2017-12-02 by Diolor