RedApparat/Fotoapparat #107: No API to query for current camera parameters

ezaquarii, 2017-09-30:
#### What are you trying to achieve or the steps to reproduce?

I want to display picture size selection menu with currently used picture size,
but the Fotoapparat API doesn't expose current camera parameters.

Those parameters are kept in `ParametersProvider` instance, that is not accessible.

#### What did you expect?

I expect information about current camera settings to be available after `start()`.

Diolor, 2017-09-30:
You can get the capabilities a camera has with [`Fotoapparat#getCapabilities()`](https://github.com/Fotoapparat/Fotoapparat/blob/master/fotoapparat/src/main/java/io/fotoapparat/Fotoapparat.java#L191). Are you looking for something more than those capabilities?

ezaquarii, 2017-09-30:
I'm not interested in capabilities, but currently selected resolution.

When camera is started with default settings, I can't tell which defaults are currently active.
I can't use capabilities to initialize the camera, because it becomes available... after you start the camera.

friendoye, 2017-10-02:
@Diolor @dmitry-zaitsev Probaly closed with #108.

closed 2017-10-02 by dmitry-zaitsev